lukevp/ESC-POS-.NET #96: Virtual Printers

dicksonkimeu, 2020-12-19:
I tested direct printing and it's working.

Does this support virtual Printers that forward the output to other printers?

lukevp, 2020-12-19:
Could you give an example of such a printer? There are several backends built in (tcp/ip, Bluetooth serial, usb serial, file system, etc.) so it would depend on how this virtual printer works. Ultimately the destination would need to be a printer that supports esc/pos or it will not print correctly. This is an esc/pos specific library, so it is not a general printing solution.

prasannaJosium, 2021-01-04:
hi, Im using windows 10, and writing an cross platform app to print on Thermal printers.
I tried the library and was able to print on an PoS printer, which connects to my PC over bluetooth as the printer gets assigned to a COM port over Bluetooth. I create a SerialPrinter(comport,buardrate) and it works. But if i connect the printer over USB (no Bluetooth) to the PC, the printer is assigned to port USB001, i believe it is a virtual port. So, how do we write to such a port? any suggestion?

igorocampos, 2021-02-02:
@prasannaJosium yes, there's a suggestion on the repo [README](https://github.com/lukevp/ESC-POS-.NET#usb-usage-guide):
[image]

lukevp, 2021-02-03:
@prasannaJosium  I’m not against OS specific USB printer support (without virtual com port). If you need this for some reason, please send us a PR with a custom UsbPrinter type! The reason I use virtual COM is that it bypasses the OS print spoolers  and you don’t have to worry about forcing the printer to be a generic device or do device discovery (Epson lets you force a COM port with the driver, so you can easily automate setting all your registers to COM3 for the receipt printer, for example.) device discovery could be useful for others though, I’m interested in supporting that too if it’s necessary for your use case.

closed 2021-02-15 by lukevp

dongfo, 2022-04-27:
it‘s good to support usb printer

jjxtra, 2022-11-21:
Is there documentation on how to do device discovery? Would be happy to help write the code and I need this for a MAUI app I am creating...

jjxtra, 2022-11-21:
Adding some info for UDP packet discovery for epson printers in case it is useful to someone.

```cs
private static readonly IPAddress encpIP = IPAddress.Parse("255.255.255.255");
private const int encpPort = 3289;
private static readonly byte[] probeMessageBytesENPC = new byte[]
{
    (byte)'E', (byte)'P', (byte)'S', (byte)'O', (byte)'N', (byte)'Q', 0x03, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x00
};

...

var udpSocket = new Socket(AddressFamily.InterNetwork, SocketType.Dgram, ProtocolType.Udp);
udpSocket.SetSocketOption(SocketOptionLevel.Socket, SocketOptionName.Broadcast, true);
var endPoint = new IPEndPoint(encpIP, encpPort);
await udpSocket.SendToAsync(probeMessageBytesENPC, SocketFlags.None, endPoint);

var timeout = TimeSpan.FromSeconds(10.0)
using var cancelTokenSource = new CancellationTokenSource(timeout);
var result = await udpSocket.ReceiveFromAsync(receiveBuffer, SocketFlags.None, endPoint, cancelTokenSource.Token);

// if response is an epson printer, include it in the list
if (result.ReceivedBytes < 1 || result.ReceivedBytes >= 8192 ||
    result.RemoteEndPoint is not IPEndPoint ipep ||
    (receiveBuffer[0] != 69 || receiveBuffer[1] != 80 || receiveBuffer[2] != 83 || receiveBuffer[3] != 79 ||
    receiveBuffer[4] != 78 || receiveBuffer[5] != 113))
{
    continue;
}

// result.RemoteEndPoint has the printer ip address, the port to print on is usually 9100, not the port in this end point
```