lukevp/ESC-POS-.NET #100: BarCode 93 not printing

pedrollanes, 2021-01-04:
Two issues: In barcode 93 I'm trying to print this code $EC0MNRN1V7G$. If I put in this value I recieve a exception that Code39 requires * at start and end, but I'm trying to use code 93. 
Second issue: I put * for avoiding this error and I have not any error but printer doesn't print the barcode.

Thanks

igorocampos, 2021-01-04:
About the error message, it is clearly a misspelling where the correct message should say `Code93` instead of `Code39`, at least I've never heard of a Code39 type of barcode. And you can check the source code that originated the message:
[image]

About the printer not printing anything, you will have to check if it is compatible with Code93 type of barcode. Which printer and model are you using? And is there any specific reason why use Code39, instead of Code128 for example?

pedrollanes, 2021-01-04:
At last the problem is that the width exceed the paper width. I have used code128 and work fine. Thanks

lukevp, 2021-02-15:
Thanks @pedrollanes for calling out the typo in the exception, this has been fixed in the prerelease version of the library and this issue is closed.  I'm glad you got the barcode to print correctly!

commit b118b98a (2021-02-15): Correct misspelling in exception

Fixes #100.
closed 2021-02-15 by lukevp