lukevp/ESC-POS-.NET #101: Print command stuck in case multiple queue printing orders

smtebooks, 2021-01-08:
Thanks for this awesome work...
My issue simply is sometimes the print order does not print to my printer specially when there is multiple orders.

for example if i put the print order in while or for loop, the app/winform is stuck and not respond at all

igorocampos, 2021-01-14:
In order to not block your UI, you should send the printing from a different thread.

About losing some content when sending multiple commands to the printer, maybe you want to implement a printing queue, so that you send the next command after you know the printer received and printed the previous one. Maybe the hardware has a buffer to receive the bytes and you are exceeding it sending too fast, too many bytes.

I think @lukevp also have implemented a buffer in this library, but I haven't look at it. You can also try to debug and check where the printing is lost.

lukevp, 2021-02-15:
Yes, there is a separate thread that flushes bytes to the printer on a schedule.  The printer library should accept bytes and store them in memory, then print them to the printer in a separate thread.  This should not be an issue as the write command just appends to a buffer in memory and does not actually wait on hardware.  @smtebooks Can you please push a repo with a minimal example of a Winforms or other UI app that has this behavior?  Would love to look into it and see if there's something missing.

As for the printer not always printing the full receipt, the printers do have a hardware receive buffer, but I'm able to stream full images across which are many thousands of bytes without filling up the buffer too badly on a real EPSON printer.  It would be almost impossible to fill the buffer with just a plain receipt unless you were printing thousands of copies, and even if you did, the printer should still allow you to print to it, it should just block while the buffer is full, but I'm not sure if this is handled correctly since I've never seen this happen personally.

lukevp, 2021-02-15:
I forgot to mention, @igorocampos's point about doing as little as possible in the UI thread is always best practice, but this library shouldn't be blocking your UI like this even if you use it this way since it uses a background thread for flushing the buffer to the underlying hardware.

closed 2021-09-05 by lukevp