lukevp/ESC-POS-.NET #103: How to print the Euro Symbol €?

atrudl, 2021-01-28:
The Euro Symbol is not available in any CodePage

lucdhaenen, 2021-02-10:
Indeed, I need it for our receipts...
Pls advise

lukevp, 2021-02-15:
Hey @atrudl  @lucdhaenen sorry for the delay.  The euro is available on several codepages.

The problem both of you are running into is that C# strings are Unicode, so you are able to declare a string with the value "€" in C#, but unfortunately, that value does not correctly map to an ASCII 8-bit value that's printable using the standard 7-bit + extended codepages.  See [here](https://www.compart.com/en/unicode/U+20AC) for more information.  This is a 2 byte Unicode string.

In order to print a Euro symbol, you must switch to a code page that supports that, then print a character with the relevant ASCII value.  Here is an example:

```csharp
private static void PrintEuro()
{
    var EURO = new byte[] { 0xD5 };
    printer.Write(ByteSplicer.Combine(
        e.CodePage(CodePage.PC858_EURO),
        EURO,
        e.CodePage(CodePage.PC437_USA_STANDARD_EUROPE_DEFAULT)));
}
```
Switching back to the standard codepage is optional. If you don't need the extended USA/Europe codepage, you could just switch to PC858_EURO and use that for your whole receipt.

I have just pushed a new build of the library that has a code page tester, option 8.  Please give it a try and print index 17 (PC858_EURO) and you will see the Euro there in the list.   Here is the example output of the PC858_EURO codepage with the Euro symbol highlighted.

[image]

I think this could be made easier to use if the PrintLine function could do CodePage switching automatically - i.e. it could accept Unicode strings and convert them to ASCII runs with CodePage switch commands embedded.  @igorocampos  do you have any thoughts on this? Feature creep or helpful functionality?

We could have PrintLine replace known unicode characters and allow it to swap codepages as necessary, but that might result in a LOT of CodePage switching potentially (up to each character, even).  Might not necessarily be an issue though.

closed 2021-02-15 by lukevp

igorocampos, 2021-02-15:
@lukevp, I don't have any function helpers that "auto-convert" from Unicode to ASCII... I think implementing the built-in mapping with CodePage switches would be the way to go. However I wonder how people actually do with Chinese/Japanese Characters, since there are thousands of characters to be mapped... Maybe a research on that would give better ideas?

igorocampos, 2021-02-15:
@lukevp I've searched a little bit about that. And perhaps we don't need to change the printer's codepage at all.
See [this](https://github.com/lukevp/ESC-POS-.NET/issues/86#issuecomment-718970421) and [this](https://github.com/lukevp/ESC-POS-.NET/issues/88#issuecomment-744974099)

If we give the user the possibility to choose the proper encoding when converting from string to byte[], they can choose whatever encoding page to use for retrieving the right byte. At least that's what I understood, especially from the second link.

Juliangzr, 2025-04-25:
instead of using Printline i created the following method:
```
    private byte[] PrintLineWithEncoding(int encodingNumber, string text)
    {
        var enconding = Encoding.GetEncoding(encodingNumber);
        return enconding.GetBytes(text + "\n");
    }
```

It  would be great if the Printline method could accept the encoding.