lukevp/ESC-POS-.NET #104: How to print a image in a different size?

atrudl, 2021-01-28:


lukevp, 2021-02-03:
Hi @atrudl what is your issue? The test program demonstrates printing images at different sizes.

atrudl, 2021-02-03:
I want to print a image in a different size but i can only define max_width as parameter. So i have to resize the image before printing

lukevp, 2021-02-03:
What situation is this not suitable? For example, if I have a 1000x500 logo and I want it to be half sized, passing a max width of 500 will result in an image of 500x250. This is an aspect ratio preserving resize. The only thing this misses is upscaling images that are too small, but this is unlikely to be an issue as receipt printers can only print at about 500px.

atrudl, 2021-02-03:
I had the problem with image that are too small, but if i resize before, it is not a problem. Thanks

lukevp, 2021-02-03:
Making small images larger is not the best approach because it will not be as clear as a correctly sized image. I’d suggest getting bigger images or using them at their native resolution. This isn’t something I feel we should support in the library. Max width makes sense because it is needed to prevent images from not fitting on receipts, but enlarging small images starts us down the path of additional image manipulation.

closed 2021-02-03 by lukevp