lukevp/ESC-POS-.NET #107: Network printer interface freezes when unplugged

RobNL, 2021-02-09:
Case 1:
A .net core 3.1 app, where I use ESCPOS_NET
Printer is accessible on an IP address
Create a new instance of network printer.
Start monitoring.
Unplug it by pulling the LAN cable or shutting down.

Result 1:
The app seems to freeze...

Result 2:
In one test, the app received and transmitted data to the disconnected printer. Then that status was changed to offline.
How can it send data to a disconnected printer?

What I would expected to happen:
-On a write attempt, the connection is checked and the status is changed to offline.
-Not freezing, but a retry for connecting every now and then (with sleep between tries)

Case 2:

Printer is NOT accessible on an IP address because I've pulled the plug
Create a new instance of network printer, with retryontimeout=true.
The app freezes, doesn't start up. Maybe an infinite loop?

Excepted:
- I'd expect it to start with a status: online=false
- I'd expect the app to retry on a regular interval, say 1 s, and reconnects after I've plugged the printer back in.

lukevp, 2021-02-15:
Hey @RobNL, the network printing was all set up by @kodejack  I believe, I don't have a networked printer unfortunately.  Can you run the app in debug mode and reproduce these with stepping turned on to see if you can figure out where the issue is?

My guess is that this is probably where the issue is.  This code looks like it attempts to reconnect and then if it isn't connected, it will set the timeout to disconnected and invoke a status change (which is, as you say, an infinite loop).  There is no pause between this, so it looks like it will invoke it over and over.

[image]

It would be good to add some delay to this to defer to the background thread.  Something like this (you'll have to test, but this will probably fix the freezing and infinite looping.  It may never reconnect though).
```csharp
if (!IsConnected)
{
    Status = new PrinterStatusEventArgs()
    {
        DeviceConnectionTimeout = true,
    };
    Task.Run(() =>
    {
        Task.Delay(1000);
        StatusChanged?.Invoke(this, Status);
    });
}
```

It would be good to also fix the Reconnect event to successfully reconnect to your printer.

lukevp, 2021-02-15:
In reference to your question of how the app can send data to a disconnected printer, you will have to look at the underlying .NET Core network stack.  A lot of times, if a device is unplugged, it doesn't send a connection close event on the TCP connection, so there is no way to tell it is not listening until a message is sent that doesn't receive an ACK.  So there are definitely situations where data can be sent to the printer and it turns out it's already not listening.  There's not much that can be done about that, but ideally you'd have some type of retry buffer and not clear an item out of it until it's successfully written, so we could implement something like that.

RobNL, 2021-02-19:
Thanks for your reply.
I've already taken your code as a basis and made a new class, where I've added some stuff to make it more responsive, because the socket.connect blocks the application for 2 minutes.

The socket class is extremely funky and the printer adds troubles of its own so it's difficult to make something that works well.
So far I've got this, which works fairly well a.t.m. although it's not a finished product and not tested in a production environment.

I'll share it because there's nothing secret here and I pulled some extra tricks from fora.

using System;
using System.Collections.Generic;
using System.IO;
using System.Linq;
using System.Net;
using System.Net.Sockets;
using System.Text;
using System.Threading.Tasks;
using ESCPOS_NET;

/*
 
A socket:
To connect to an offline printer, the sockect.connect has a 90 seconds timeout and blocks the application.
--> remedy: the ASYNC connect is used.

The printer also has a timeout. It'll forcibly close the socket from the printer's end, if there's no write attempt in 90 seconds.
--> remedy: a KeepAlive thread with a simple write command.

The socket has a Poll that only returns its internal state. It doesn't have a 'ping'.
--> remedy: do a PING before trying to connect. Also do 

The socket stays alive and connected and writes data, even if the cable is unplugged.
--> remedy: do a PING before a write.

Another case:
The LAN cable to the printer is unplugged for more than 90 seconds.
The connection is then plugged back in.
The printer has focibly closed the connection
A write will fail
--> remedy: on write fail, retry 1x to reconnect automaticallu.

*/

namespace Printer.Lib
{

	public enum PrinterErrors
	{
		none = 0,
		connection_could_not_be_made = 1,
		connection_fail = 2,
		write_try1_failed = 4,
		write_try2_failed = 5,
		keepAlive_fail = 6,
		connection_was_lost = 7,
		ping_nosuccess = 8,
	}

	public class PrinterStatus
	{
		public bool isConnected = false;

		public PrinterErrors errorCode = PrinterErrors.none;
		public string errorMessage = "";
		public string warningMessage = "";

		public bool HasError()
		{
			return this.errorCode != PrinterErrors.none;
		}

[... 592 lines cut ...]
					Console.WriteLine(string.Format("read exception: {0}", ex.Message));
				}
				catch (Exception ex)
				{
					// swallow the exception
					Console.WriteLine(string.Format("read exception: {0}", ex.Message));
				}

				Console.WriteLine(string.Format("ReadAll on {0}:{1}", this.host, this.port));

			}

		}

		protected override void OverridableDispose()
		{
			this.sockStream?.Close();
			this.sockStream?.Dispose();
			this.socket?.Close();
			this.socket?.Dispose();
		}

	}

}

RobNL, 2021-02-19:
Actually the KeepAlive thread doesn't work well, so you can ignore that part...

lukevp, 2021-09-05:
Hey @RobNL!  Since we last spoke, I purchased both an Ethernet as well as a Wi-Fi printer.  I did a major deep dive around the networking code in ESCPOS_NET and have completely rewritten it.  The latest master of ESCPOS has been bumped a major version to account for some breaking changes.

I would love it if you'd give it a try and provide feedback!  The latest version seems to work no matter what you do - unplugging the printer halfway through a print, opening/closing the print door, jamming, disconnecting wifi, whatever... Sometimes it does take a little while to determine the printer is no longer connected, but other than that, it should be considerably more robust now.

Looking forward to your feedback!

fatihbabacan92, 2021-09-23:
@lukevp 

I have written a lot of hacky bypasses to make the network part work seamlessly, on an older version of your library though. 

On Windows/UWP it works perfect, especially with USB. But the network printer sometimes seems to congest and get disabled for two minutes. There's also a 90 second timeout after connecting to the printer. If you don't send a print request within those seconds, the printer also gets disabled for two minutes. I worked around that by only connecting when we have something to send (after polling for incoming orders every 10s), and then I dispose the printer via printer.dispose(). I also had rewritten some parts within your library to prevent socket crashes.

I do still have problems with Android devices going to the background. Sometimes it doesn't want to connect for two minutes or a restart is required when our app switches from foreground to background. We stop the printing service for the foreground and then start the printing service for the background.

I have a deadline to make this work, but I'll surely update to the latest version next week and test it out. I'll create a PR for the remaining problems you may have overlooked in the new version.

lukevp, 2021-09-24:
Thanks @fatihbabacan92 , glad to hear you are using the library and the USB part works well!  I developed the library originally on Windows with USB, so that's certainly why that's the most stable interface 🤔 

This version is significantly more performant with the networking.  I'm very interested to hear your feedback.  It's a ground-up rewrite so I wouldn't worry about your past experience and give this a try. 

You should also no longer need to connect on-demand.  You can create one Printer object and keep it for the lifetime of your application, and it will automatically connect as soon as the printer comes online and fire events anytime the printer changes status.  The 90 second timeout is not a native thing to sockets or the library, that's implemented on the printer side (it drops connections after x seconds if you have it configured that way).  I'd suggest you change that in the printer settings - if you have an Epson, you can use EpsonNet to disable the TCP connection timeout.  This new version will still work in that scenario, it will just be re-connecting every 90 seconds.

The main reason why you'd want to do this is that when the printer connects, it enables automatic status back, so you can get events anytime the printer changes states right away.  If you don't keep the persistent connection, you won't know if a print will actually go through when it comes time to print.  

If you do want to print on-demand anyway, and are OK losing the status messages, you can create and dispose of printers for each print, but I haven't tested this use case.

Regarding Android devices, I haven't tested in that scenario.  How are you using it on Android?  Is it in a Xamarin app?  My guess is going in the background causes a lot of the timers to get messed up since backgrounding on mobile typically throttles the code significantly.  however, the auto-connect and health-check logic in the new version `may` just work.  I'd love to hear if you have issues with Android in the new version and I can set up a similar test to give it a try.

fatihbabacan92, 2021-09-24:
@lukevp First of all, I already quickly migrated to the new version. It works marvelous without having to change our code too much.

We still use the dispose method. We only need the status to check if the drawer is open or if the paper is out. So it works out greatly for us. What we do is API Call Order -> if orders > 0 -> if !isPrinterOnline -> Connect -> if !order.printed -> print order -> if isPrinterOnline -> mark order as printed -> dispose. After printing the order we check if the printer is still online, that way we know if it was printed or not :)

Yes, we are using Xamarin for UWP, iOS and Android. Your library even works on MacOS with Avalonia. The issue we had was just a conflict by not properly destroying the Foreground Service. When the App pauses we dispose the printer and reconnect via the Foreground Service.

Thank you for the great work Luke! If you want more info about Xamarin or whenever someone here needs help about it, tag me.

lukevp, 2021-09-24:
That's awesome feedback @fatihbabacan92 , I'm glad the new version is working well!  That's really great that you're using it cross-platform.  If you do come across any issues, please open an issue so we can discuss, I'll close this one since both of us have validated the new version's networking functionality.  There is a separate story that's been open a while for Xamarin, so I'll tag you over there.

closed 2021-09-24 by lukevp