lukevp/ESC-POS-.NET #111: Printing Images - Height

Gi4nfratti, 2021-04-27:
Hello, I made a [question](https://stackoverflow.com/questions/67288683/print-image-esc-pos) on Stackoverflow (for more details), about the height of the image when I print. I'm using Xamarin Android, I created a bitmap with SkiaSharp and save it as PNG. It's perfectly saved on my device. But when I'll print it, the necessary height to print the whole bitmap is not matching. I tried with [this](https://logosmarcas.net/wp-content/uploads/2020/04/Nike-Logo.png) image, it's a logo with a width more larger than the height, 3840x2160, and it worked perfectly. And my bitmap have the height more larger than the width (it's a list), 700 of width and 1280 of height. Both images are PNG and 72 of DPI. I'm reducing the width as follows:

` buffer = ByteSplicer.Combine( e.PrintImage(logo, true, true, 372, 0) );`

If I don't reduce the width, it prints a lot of random characters.
The width is perfect, but I have the height similar to the first image, it seems a "logo size".

Can you show an example of how can we print images/bitmaps that have a big height like a list without cutting/limiting it?

lukevp, 2021-09-05:
Hi, sorry for the delay getting back to you.  What happens if you set the width on the image and print it?  Are there other too-tall images that print correctly?  The reason I ask is that the bar prints the whole way down, so that makes me think the problem might be with the PNG you are using and not with the lib resizing the image.

Gi4nfratti, 2021-09-05:
Hello, no problem. This was resolved by dividing the bitmap in parts of 256px of height and then printing.

closed 2021-09-05 by Gi4nfratti

jjxtra, 2022-11-18:
Had to do something similar, sharing for future generations...

```cs
    private static IEnumerable<Image<Rgba32>> ExtractChunks(Image<Rgba32> sourceImage)
    {
        const int chunkSize = 256;
        for (var y = 0; y < sourceImage.Height; y += chunkSize)
        {
            Rectangle sourceArea = new(0, y, sourceImage.Width, Math.Min(sourceImage.Height - y, chunkSize));
            if (sourceArea.Height < 1)
            {
                yield break;
            }
            Image<Rgba32> targetImage = new(sourceArea.Width, sourceArea.Height);
            int height = sourceArea.Height;
            sourceImage.ProcessPixelRows(targetImage, (sourceAccessor, targetAccessor) =>
            {
                for (int i = 0; i < height; i++)
                {
                    Span<Rgba32> sourceRow = sourceAccessor.GetRowSpan(sourceArea.Y + i);
                    Span<Rgba32> targetRow = targetAccessor.GetRowSpan(i);

                    sourceRow.Slice(sourceArea.X, sourceArea.Width).CopyTo(targetRow);
                }
            });
            yield return targetImage;
        }
    }
```

nivle, 2023-06-13:
I ended up doing solving this like this.....(to solve some buffer issue), which is kind of the same as setting legacymode to true I guess)... though in this way you have some more control over the size...

```

IList<byte[]> data = new List<byte[]>
            {
                epson.CenterAlign()
            };

Image image = Image.FromStream("///");//add here your image

int size = 256;

for (int i = 0; i < Math.Round((double)image.Height / size, 0, MidpointRounding.ToPositiveInfinity); i++)
{
    int width = image.Width;
    int height = size * (i + 1) <= image.Height ? size : image.Height - (size * i);

    Bitmap newImage = new Bitmap(width, height);

    using (Graphics graphics = Graphics.FromImage(newImage))
    {
        graphics.DrawImage(image, new Rectangle(0, 0, width, height), new Rectangle(0, i * size, width, height), GraphicsUnit.Pixel);
    }

    using (MemoryStream memoryStream = new MemoryStream())
    {
        newImage.Save(memoryStream, ImageFormat.Tiff);
        data.Add(epson.PrintImage(memoryStream.ToArray(), true));
    }
}

data.Add(epson.FullCutAfterFeed(0));

return ByteSplicer.Combine(data.ToArray());

```