lukevp/ESC-POS-.NET #113: Possible support on using installed printer name?

ferrywlto, 2021-06-05:
Thank you for such a great library. When using it on Windows platform with a USB/Network connected printer, its easy to access the printer via virtual COM port. However when using the library on a Mac (my main dev machine). The `/dev/ttyusb*` serial file is not always appear (I dunno why). But I can still access the printer via `PrintDocument` class and `PrinterSettings.InstalledPrinters`. Is there any plan to support this or directions on how could I implement this so I could contribute a PR then?

lukevp, 2021-06-05:
Hi @ferrywlto , glad you like the library! We would love it if you could contribute.

printing on Windows is kind of finicky with USB drivers, so I opted to go the virtual COM route. I’m open to having an explicit USB version that requires drivers as well. 

can you do this with .net dependencies only, or will it require another package?

Please feel free to take a stab at it and open a PR. Some things I’d like to see would be some unit tests, console tests, a check that the platform is OSX, and verifying that the printing is properly happening in RAW mode and not TEXT mode. RAW mode lets you send a binary stream. If the printing is TEXT mode, it will appear to work with Print commands but others like images will not work correctly.

ferrywlto, 2021-06-06:
Thanks @lukevp, I tried to use the .net5 built-in ` System.Drawing.Printing.PrintDocument` class and able to connect my EPSON TM-m10 printer installed on Mac. However it feel likes drawing on a canvas and print to printer as a file. (Act like what a desktop printer do.) Which defeat the purpose of using ESCPOS command.  Therefore I don't think it will be easy to make this library support the installed printer. 😥

lukevp, 2021-06-06:
That's too bad.  Sounds like it's a rasterized printing interface, so that wouldn't work with the library.  What printer are you using?  Perhaps we can figure out why it is not reliably showing under the /dev/ttyusb .  Are there any factors that make this happen? Eg. it disappears after reboot, or after a certain # of minutes, or what?

ferrywlto, 2021-06-07:
I am using EPSON TM-m10 thermal printer. It was strange that the first time I plug in the printer I can access it via /dev/ttyusb. However after some test prints using the library /dev/ttyusb had gone completely for unknown reason. It shouldn't due to the library as switching USB port and even another printer the /dev/ttyusb didn't work either.

By the way I found that there is a 1 cm height empty spaces at the top of each print out, is there any way to reduce it?

ferrywlto, 2021-06-12:
Closing this issue as technically infeasible.

closed 2021-06-12 by ferrywlto