lukevp/ESC-POS-.NET #115: Font size for Chinese words

jajazhou, 2021-06-14:
i try to set the font style as :
e.SetStyles( PrintStyle.DoubleHeight | PrintStyle.DoubleWidth ),

the words include English & Chinese words, but the Chinese words cannot show as double Hight & double width, any solution?

[image]

linycyc, 2021-06-14:
hi jajazhou
I want to ask you how to print the tradition chinese word ?
I try to encode utf-8 and big5, but not working, i use virtual printer form RawBT app

Thanks~

jajazhou, 2021-06-15:
> hi jajazhou
> I want to ask you how to print the tradition chinese word ?
> I try to encode utf-8 and big5, but not working, i use virtual printer form RawBT app
> 
> Thanks~

@linycyc 
changed this function:
        public virtual byte[] Print(string data)
        {
            // Fix OSX or Windows-style newlines
            data = data.Replace("\r\n", "\n");
            data = data.Replace("\r", "\n");

            // TODO: Sanitize...
            return data.ToCharArray().Select(x => (byte)x).ToArray();
        }
to:

        public virtual byte[] Print(string data)
        {
            // Fix OSX or Windows-style newlines
            data = data.Replace("\r\n", "\n");
            data = data.Replace("\r", "\n");

            // TODO: Sanitize...
            return Encoding.GetEncoding("BIG5").GetBytes(data);
        }

jajazhou, 2021-06-15:
I solved this problem myself,
add the function SetFonSize to class BaseCommandEmitter,

`public virtual byte[] SetFonSize(FontStyle style) => new byte[] {  Cmd.GS, Chars.StyleMode, (byte)style };`

and the FontStyle enum as :

    [Flags]
    public enum FontStyle
    {
        None         = 0,
        DoubleHeight = 1,
        DoubleWidth  = 1 << 4 
    }

print example:
`            
```
             ..........

              e.SetFonSize( FontStyle.None), // reset to normal size

              e.SetFonSize( FontStyle.DoubleWidth), // set to double width

              e.PrintLine(somewords),

              e.SetFonSize( FontStyle.None),`  // reset to normal size

              ..........
```

linycyc, 2021-06-15:
Hi @jajazhou 

I try to use the new function, but the RawBT app still not show traditional chinese
whether i need to use real pos printer device for my test?

```
  public virtual byte[] Print(string data)
    {
        // Fix OSX or Windows-style newlines
        data = data.Replace("\r\n", "\n");
        data = data.Replace("\r", "\n");

        // TODO: Sanitize...
        return Encoding.GetEncoding("BIG5").GetBytes(data);
    }
```

jajazhou, 2021-06-16:
> Hi @jajazhou
>
> I try to use the new function, but the RawBT app still not show traditional chinese
> whether i need to use real pos printer device for my test?
>
> ```
>   public virtual byte[] Print(string [...]

Yes, i think so. your printer must be supported for ESC/POS standard by Epson.

lukevp, 2021-09-05:
@jajazhou Would you be interested in contributing the FontSize changes as a PR?  Would love to get you registered as a contributor in GitHub.  If not, I can add support for this myself.  Appreciate the contribution!