lukevp/ESC-POS-.NET #116: Print for Traditional Chinese

linycyc, 2021-06-14:
How to print the traditional chinese word from PrintLine function ?
I try to encode utf-8 and big5, but not working, and i use the virtual printer from RawBT android app to test

lukevp, 2021-09-05:
You can only print simplified Asian alphabets (for example, in Japanese, you can only print katakana and hiragana and the kanji support is limited, although there are code pages for it).  The cause of the limitation is that it must have character pages available in your printer that you can switch to that represent each character (it doesn't support unicode strings directly).  

Now, if you live in China you will probably have a different set of code pages than what the US market has.  Do you have a manual for the specific printer you have?  That would be one possibility for printing your unicode strings.  If you want to print traditional Chinese characters in a way that works across all printer models, the way to solve that would be to either

1. Create a dynamic character map in memory on the printer and stream each character into the character map as you print, or
2. Rasterize the unicode text and print it to the printer as an image.

This is something that would be super cool to support, so if you're interested to deep dive into either, please let me know and I will be happy to guide you on implementing this - although I will not be able to implement this myself.

dredgy, 2022-01-07:
Hi Luke, I’m super interested to help implement the rasterisation method. Did you ever get very far with looking into it?

igorocampos, 2024-01-23:
Related to Original Question: https://github.com/lukevp/ESC-POS-.NET/issues/88#issuecomment-744974099