lukevp/ESC-POS-.NET #119: QRCode help

Dufayel, 2021-08-12:
Hi, there is any way to print a qrcode with left text? like that image bellow?
[image]

igorocampos, 2021-09-03:
Hello there, I believe the only way to accomplish that is using the PAGE MODE. Unfortunately I don't think it has been implemented in this library. But you are very welcome to contribute :) 

[image]

Since you are brazilian, [here](https://www.projetoacbr.com.br/forum/topic/43887-qrcode-e-logotipo-lateral-em-danfce-escpos/) is a forum that had it implemented in their open source solution. You can easily check their source code for the necessary commands and implement through a PR in this project.

Below a list of commands that would more likely to be implemented:
- `ESC L` -   Select page mode 
- `ESC FF`  - Print data in page mode 
- `ESC T` -  Select print direction in page mode
- `GS $` -  Set absolute vertical print position in page mode 
- `GS  ＼`-  Set relative vertical print position in page mode
- `ESC W` -  Set printing area in page mode

Dufayel, 2021-09-03:
Thank you very much! it's saved my life.

closed 2021-09-03 by Dufayel