lukevp/ESC-POS-.NET #122: No Barcode when printing from Linux

BenderCanCode, 2021-08-31:
I am trying print a barcode in Linux/Raspbian, but no luck it prints blank at the spot, works perfect in Windows. I don't know if it is the printer that issue the issue.

igorocampos, 2021-09-02:
Does it print everything else in Linux? The issue is only for barcode?

BenderCanCode, 2021-09-03:
Yes, so I am using the FilePrinter method. and I don't have drivers installed.

igorocampos, 2021-09-03:
It shouldn't need them on Linux, from what says the README.md

> On Linux and Mac, USB for Epson printers is exposed as a serial port directly by the os under /dev/ttyusb or something similar based on your platform, and doesn't require drivers.

Very odd that all the other commands work just fine... That means the printer **is** receiving the bytes.
Well, I'm no linux expert, but in your situation I would try to share the printer and send the commands (as a file, like FilePrinter command) to its shared address. For instance, I've done that in windows before like this:
`copy commands_file.tmp > \\192.168.0.100\printer`

lukevp, 2021-09-05:
@BenderCanCode sounds like you are interacting with the printer in some type of non-raw mode, so it's affecting the byte stream and not passing it directly.  Do any other ESCPOS commands work (bolding, centering text, etc.) 

What file path are you writing to on Linux, what version of ESCPOS_NET, and how do you have the printer physically connected?

BenderCanCode, 2021-09-06:
Every other styles works when I used this file path "/dev/usb/lp0", it is connected via USB directly on to the raspberry pi unit.

BenderCanCode, 2021-09-06:
I figured out what the problem was, When I was testing on windows, I was using ITF Barcode Type, when I was printing to linux I set it to Code128, and it printed blank for the barcode, I checked the printer specs and saw it wasn't compatible with the print. Thanks for the assistance.

closed 2021-09-06 by BenderCanCode