lukevp/ESC-POS-.NET #123: Is it possible to get printer char width?

maximuss, 2021-09-05:
Is it possible to get printer total printable width in chars?
I can't seem to find that info anywhere.

I have been doing some research and I have noticed that in Epson's OPOS for .NET they are using this command:

```
int iRecLineCharsCount = m_Printer.RecLineCharsList.Length;
```
Where m_Printer is instanced this way:

```
//Create PosExplorer
PosExplorer posExplorer = new PosExplorer();

DeviceInfo deviceInfo = null;

try
{
	deviceInfo = posExplorer.GetDevice(DeviceType.PosPrinter,strLogicalName);
	m_Printer =(PosPrinter)posExplorer.CreateInstance(deviceInfo);
}
catch(Exception)
{
	ChangeButtonStatus();
	return;
}
```
You can find the program to install here:
https://download.epson-biz.com/modules/pos/index.php?page=soft&scat=39

If you only install the examples you can find the code in the "PrinterSample_Step7 in file FrameStep7.cs

Best Regards
Bjarne Pedersen

lukevp, 2021-09-05:
Hey @maximuss thanks for reaching out and providing this detail!

I haven't used OPOS for anything serious, but I will tell you how I **think** this works, and perhaps you can do some more research and let us know what you find.

My understanding is that OPOS (and UPOS) are operating at a level above the ESC/POS protocol.  They do ESC/POS when interacting with the devices, but they also provide a uniform interface to communicate with devices.  So when you install a device driver that has support for OPOS/UPOS, you are also registering a device with your OS within the OPOS/UPOS system. 

So really when you query DeviceInfo what you are doing is extracting the information that the driver provider registered with OPOS.

ESCPOS_NET as a library operates directly via serial/TCP, so it doesn't have knowledge of the devices, drivers, or anything like that, it only knows that a device is or isn't listening on a given port or not.  This is one of the advantages of ESCPOS_NET - it is cross-platform to MacOS, Windows, and Linux, and works across USB, Bluetooth, WiFi, Ethernet, Serial, and Parallel - because it doesn't use device drivers and speaks ESC/POS directly. 

If you are interested in taking this further, the way that we could accomplish this would be to use the Printer ID command to get the printer model, and then provide a lookup table of printer metadata within the ESC/POS library.  I bet we could find a table or some other information online that would help fill this table out.
https://www.b4x.com/android/forum/threads/how-to-find-out-paper-width-for-an-esc-pos-printer.123231/

@igorocampos do you happen to have any experience with OPOS?

igorocampos, 2021-09-06:
Hi @lukevp this is actually the first time I actually heard of OPOS / UPOS ...
I don't think it's really used down here in Brazil, everyone I've came across just go for plain ESC/POS.

maximuss, 2021-09-06:
Hi @lukevp 
I think you are right about your assumption about how it work. I had a hunch about it.

So I have tried to find some info about my Epson printer (TM-T20III) and I have found this:
"Column capacity receipt -  Paper width 80 mm, 48 / 64"

From this site:
https://www.epson.eu/products/sd/pos-printer/epson-tm-t20iii-series#specifications

igorocampos, 2021-09-07:
FYI, one workaround that I've seen out there is to create a get only property on the instance class that set by a required parameter in the ctor which will determine the total of columns a line has.

And I believe it is that way because the paper roll can also vary, right? 

So they set the columns based on the current paper roll.

lukevp, 2021-09-16:
Yeah, you can adjust the character width by changing the font, and the paper width can also vary (some printers can print to multiple paper widths and some printers have smaller rolls by default) so it's something that would have to be read from metadata.  Would be cool to have a lookup table of printers and their corresponding print widths (and other attributes).  If you're interested in starting this @maximuss  would love a PR.