lukevp/ESC-POS-.NET #124: Slow print after upgrade

neon-dev, 2021-09-08:
A user reported slow prints (SerialPrinter) after updating to 2.0. Is there some new setting I need to define?
Thanks

neon-dev, 2021-09-09:
I found the reason:
https://github.com/lukevp/ESC-POS-.NET/blob/a5e3fb8f2ad027dd80d90ea4b7886ecd9c79f3f4/ESCPOS_NET/Printers/BasePrinter.cs#L100
Since our data is sent line by line, this delay slows printing down dramatically.

igorocampos, 2021-09-09:
Few more info about this change. It was introduced at https://github.com/lukevp/ESC-POS-.NET/commit/9a68c53354beabc6593627649398d3759b9320ab and there was a Delay of `500` ms before in `WriteLongRunningTask` method, but in a different line.
[image]

So the said commit actually moved it up.

neon-dev, 2021-09-09:
Yes and the 50 ms delay got introduced with 32fae5e8 but never got published with any version it seems.
v1.6.0 and 1.4.0 (the two most recent versions before 2.0) use the same synchronous code:
https://github.com/lukevp/ESC-POS-.NET/blob/ec3283954b0b19f99db33a82cadbfb5d915c10c5/ESCPOS_NET/Printers/BasePrinter.cs#L97-L122

igorocampos, 2021-09-10:
@lukevp do you know why there was a double in the delays?
From commit description is says `Refactor delays in main loop so that it's delayed even if things are null`

lukevp, 2021-09-16:
@igorocampos  @neon-dev I'll take a look this weekend and factor this in to the process.  Sorry for the issues!  I versioned it up to 2.0 since there were major changes to the networking and threading code in this release.  Thanks for the PR @neon-dev, I'll probably take in the fix and tweak it slightly.  The purpose of the sleeps overall is so that each read/write thread doesn't busy loop (since they are in a tight while loop that never ends, it will peg a CPU core at 100% if there is no deferring or async actions taking place).  If there is actually work to be done, there shouldn't be a need for a delay.  I'll provide more comments on the PR when I review.

neon-dev, 2024-07-30:
Hi @lukevp, after staying on v1.4.0 for years due to this issue I've now had to update to the latest version for the QR code fixes.
Therefore I implemented this workaround for the slow print issue:
```cs
internal class FastSerialPrinter : SerialPrinter
{

    private readonly BlockingCollection<byte[]> queue;

    public FastSerialPrinter(string portName, int baudRate) : base(portName, baudRate)
    {
        queue = new BlockingCollection<byte[]>(WriteBuffer);
    }

    public override void Write(byte[] bytes)
    {
        queue.Add(bytes);
    }

    protected override async void WriteLongRunningTask()
    {
        var field = typeof(BasePrinter).GetField("_writeCancellationTokenSource", BindingFlags.Instance | BindingFlags.NonPublic);
        CancellationTokenSource _writeCancellationTokenSource = (CancellationTokenSource) field.GetValue(this);
        TimeSpan waitTime = TimeSpan.FromMilliseconds(200);
        while (!_writeCancellationTokenSource.IsCancellationRequested) {
            if (!IsConnected) {
                await Task.Delay(waitTime);
                continue;
            }

            try {
                if (queue.TryTake(out var result, waitTime) && result != null && result.Length != 0) {
                    WriteToBinaryWriter(result);
                }
            } catch {
            }
        }
        Logging.Logger?.LogDebug("[{Function}]:[{PrinterName}] Write Long-Running Task Cancellation was requested.", $"{this}.{MethodBase.GetCurrentMethod().Name}", PrinterName);
    }
}
```
Since I have no time to update #125, I'll close the PR and hope you can take this code as a proof of concept for a final fix in the BasePrinter class.
I did not observe slowdown issues with the `ReadBuffer` yet, but maybe it would make sense to also wrap it in a `BlockingCollection`.