lukevp/ESC-POS-.NET #126: It is supported image print using RAW format on Linux OS?

Rombersoft, 2021-09-13:
I've gotten such error on Linux:
https://github.com/mtmsuhail/ESC-POS-USB-NET/issues/17

lukevp, 2021-09-16:
Hi @Rombersoft  thanks for bringing this to my attention.  I'm going to close this as a duplicate and reopen the other issue.  Let's talk there.

lukevp, 2021-09-16:
Duplicate of #17

lukevp, 2021-09-16:
Whoops, I just realized that you linked this issue to a different library, not this one.  @Rombersoft could you confirm which nuget package you're using and which version you're seeing a problem with, which OS and receipt printer, and how you are connecting to it? Could you provide a snippet that reproduces the issue on Linux?  This library was developed and tested on Linux and MacOS but I haven't re-validated the functionality recently so this would help me give it a try.

Rombersoft, 2021-09-21:
I did not try your library up to this topic. I've just ask about RAW format. Now I trying your library and printer does not print.
I have CUSTOM VKP80 thermal printer. I use Debian 10 OS.
My code:
```
var printer = new FilePrinter("/dev/usb/lp1");
var e = new EPSON();
printer.Write(ByteSplicer.Combine(e.CenterAlign(), e.PrintImage(File.ReadAllBytes("/home/destructor/Images/BlackRect.png"),true), e.PrintLine("")));
```
But nothing happens.