lukevp/ESC-POS-.NET #128: Is it possible to print it to pdf

poojaDassani, 2021-09-15:
Can i make a pdf of it and print to A4 size paper

lukevp, 2021-09-16:
Hey @poojaDassani,

Could you provide more info on your use case?  Do you want to save the receipt for later printing?  This library is specifically for interacting with thermal printers, so it wouldn't really help with printing receipts to PDF - that would need to happen separately by using something like wkhtml2pdf or some other approach for templatizing PDF generation.

poojaDassani, 2021-09-16:
Hey @lukevp  

Thank you for the reply. As the library is for thermal printer , is there any other library you have known that could work with A4 printers or any library that supports silent printing in uwp for printers other than thermal printer.

igorocampos, 2021-09-20:
@poojaDassani it sounds like you don't want to print stuff using ESC/POS, right? I mean, I don't think there are A4 printers that support ESC/POS, maybe I'm wrong here, but even if there are, I bet it wouldn't be as many as thermal printers. Then I'd say this project won't really fit what you want. You'd need to search a little bit in another direction.

I think I can give you some head start. 

About the PDF, I recommend [iText 7](https://www.nuget.org/packages/itext7/), This library is very popular PDF tool for Java and C#, which means wide community support. And also, they have a double license which allows you to either have it for free under AGPL and not commercialize your solution, or buy a license (there are many license purchase models) and do whatever you want.

About silent printing, I'm not really into UWP, but what I've successfully done in the past in WinForms is to send the text directly to the printer's port e.g. `LPT1`, `COM3`, etc. Or even to its network shared address, e.g. `\\192.168.1.152\printer`
You can even test that using command line, something like `DIR > COM3` or `DIR > \\192.168.1.152\printer`, or create a file, copy it and paste into the port/address. Anyway, hope this give you a kickstart.