lukevp/ESC-POS-.NET #129: Print QR

djliterat1984, 2021-09-20:
https://client.poscariloom.com/public/offline?request=eyJpZFVzdWFyaW8iOiJkZW1vcGRzQGNvbnRhYmlsaXVtLmNvbSIsInJlZkV4dGVybmEiOiI4ODFjNmE3Mi1hYmMwLTQ5MDctYmE5Mi0wYzA1YjJhMzMzMWEifQ==

When I pass this data to the PrintQRCode() this only print the number of the selected QRModel:

if I select QRCODE_MODEL2, print "2", if I select QRCODE_MODEL1, print "1". Maybe it's for the length of the URL? I try in Android app, and pass this URL and it prints excellent. Any idea why this happens?

I see in other issue, solution that if you modify the length of the data string in the project code, after clone this, need to function ok. but I did this and anything happen.

thanks!

igorocampos, 2021-09-23:
A few questions for you:
- Have you tried to run the Console Tests ? It has an option to test QRCodes
- If you use another string as QRCode data, does it print correctly?
- What printer model are you using? Does it support QRCode ESC/POS commands?
- Can you provide the full code you are using ?

djliterat1984, 2021-09-27:
Hi @igorocampos. 

- Have you tried to run the Console Tests ? It has an option to test QRCodes

**Yes I Tried, and I don't have any case approved**

- If you use another string as QRCode data, does it print correctly?

**Yes, when I pass short string, this print perfectly. I have Thermal Printer that not print anything, only the number of the QRModel, and in other Thermal Printer is not Printing the QR completely. Another to this, I downloaded an App, to my tablet, and I add the  long QrData and Print Excellent**

- What printer model are you using? Does it support QRCode ESC/POS commands?

**Basically this printer doesn't model. But I attach the info of this Printer. It supports QRCode ESC/POS commands**

- Can you provide the full code you are using 

**
Here I get the URL to create the QR.
       public static string GetUrl(string userId, string cartId)
        {
            var baseUrl = Consts.BaseOfflineReceiptUrl;
            var args = GetParams(userId, cartId);
            string content = JsonConvert.SerializeObject(args);
            var bytes = Encoding.UTF8.GetBytes(content);
            var encodeContent = Convert.ToBase64String(bytes);
            var url = baseUrl + encodeContent;

            return url;
        }
        
        var url = GetUrl( userId, cartId);
        e.PrintQRCode(url, TwoDimensionCodeType.QRCODE_MODEL2, Size2DCode.LARGE, CorrectionLevel2DCode.PERCENT_30),
**
[image]

Thanks!

igorocampos, 2021-09-27:
Reading what you said, sounds like we truly have a bug, but I can't identify where. Are you able to post here the said "Technical Manual of Portable Receipt Printer" for details? More specifically what that manual says about print QRCode command?

djliterat1984, 2021-09-30:
I look for "Technical Manual of Portable Receipt Printer" in google, but i don't find anything. I find only this (the attached files), but I think that you know the
[image]
[image]
m.

lukevp, 2021-10-01:
Does it work if you use PrintQRCode(url) without the additional parameters?  Maybe one of the parameters you are setting does not work.

djliterat1984, 2021-10-01:
No. It doesn't work either.

igorocampos, 2021-10-01:
[image]
Model 2 should be able to receive up 4296.

I'm pretty sure we had someone testing pretty big strings 2k+ big. I can look for the issue number about it, but perhaps that 58 cap is on the printer you are using?

igorocampos, 2021-10-01:
Here it is
https://github.com/lukevp/ESC-POS-.NET/issues/71#issuecomment-701730942

djliterat1984, 2021-10-04:
Yes, I see this post, but I don't know why when I send 59 characters to the printer, doesn't print.

igorocampos, 2021-10-04:
I think it is a restriction imposed by the hardware rather then ESC/POS. Are you able to contact the manufactor and ask them? You should be able to provide all bytes you sent the printer for their analysis

djliterat1984, 2021-10-05:
The model of the Printer is 58HB6 - Thermal Printer. It's made in China...I don't have any contact. 

I install this App in my tablet, and when I add the string to generate the QRCode, and then print, it prints perfect. 
https://play.google.com/store/apps/details?id=com.qrmango.print&hl=en_US&gl=US.

igorocampos, 2021-10-05:
Now that is interesting! What do you think about it @lukevp ? Maybe contact that app's dev to ask for the ESC/POS command they are using? And see what's the difference? No idea of what else could be causing the cap, if not the hardware...

djliterat1984, 2021-10-06:
I was reading the ESC/POS command, and compare with yours code. Excuse me for my ignorance, what is the pL and the pH when are you setting the code data? And the d1...dK. Maybe is something like this, because in the manual of ESC/POS command they are values to set and are not fixed data.

djliterat1984, 2021-10-06:
[image]

djliterat1984, 2021-10-06:
ESC-POS-.NET/ESCPOS_NET.UnitTest/EmittersBased/EPSONTests/BarCode.cs

 in this file I say

djliterat1984, 2021-10-07:
I saw the code, with the two libraries and it's the same bytes (with a many differences) and the result is the same with my thermal printer.
I have 3 thermal printers. One of these has USB and Bluetooth connection (it's a mini portable Thermal Printer), the other two only supports USB connection (one of these is Epson Printer). 
The other library was function with the other two printers, but with the USB and Bluetooth Printer is not print the QR.
I see in some Apps to print on Android that you have select Printer Type, and these are the options: EpsonPrinter, StarPrinter and Generic Text Printer. Maybe this is the problem.