lukevp/ESC-POS-.NET #132: Font size bigger than PrintStyle.DoubleHeight | PrintStyle.DoubleWidth

djnuller, 2021-09-24:
I have a use case where I would like a number (for a queue ticket) to be printet bigger than what PrintStyle.DoubleHeight | PrintStyle.DoubleWidth has to offer. 
I'm not able to find anything about that in the documentation for ESC / POS (https://reference.epson-biz.com/modules/ref_escpos/index.php).

Do you know if its possible? Or can direct me to some better documentation, I would love to help out with this lib if I can. 

Br
Jesper

lukevp, 2021-09-24:
That's the biggest configuration you can get if you are using the native fonts in your printer.  You can also check between fontA and fontB, fontA is usually bigger.  

However, the library is quite fast as long as you're using a solid connection to the printer, and you could easily rasterize text to an image and print that way.  We have seen a LOT of feedback where people are trying to print unicode characters (chinese, japanese) and while there's a native way to do that for certain characters, if we were able to use system fonts to render to an image and then print that, it would allow a lot more use cases since it would work for any unicode characters.  I'm thinking something like PrintRasterizedTextLine(someUnicodeString) would satisfy a lot of these requests, and you could add a config object that set the fontSize, fontWeight, fontStyle, etc.

This would be a wonderful contribution to the lib and would help many people :) Let me know what other questions you have about this and I'd be happy to answer / help however I can.  Thanks for the offer of help @djnuller 🥳

djnuller, 2021-09-28:
I have _something_ working in my own code right now, not at all ready for this library. But is it okay to use an external library aswell? I have been using ImageSharp (still in beta) for my own code to have a working solution. 

` public byte[] GetImage(string text)
        {
            var memoryStream = new MemoryStream();
            Image image = new Image<Rgba32>(300, 250);
            
            image.Mutate( x=> x.DrawText(text, Font, Color.Black, new PointF(0, 0))); 

            image.SaveAsPng(memoryStream);
            
            var bytes = memoryStream.ToArray();
            memoryStream.Dispose();
            return bytes;
        }`
        
This is a bit heavy but is working for my solution.

lukevp, 2021-10-01:
@djnuller ImageSharp is already a dependency of the library so this would be no issue to add to the library actually, it would not add any additional overhead. 
[image]
Please do open a PR, and to add to my recommendation from ☝🏼 
I'm thinking something like `PrintRasterizedTextLine(string inputText, RasterizedPrintOptions options)`
The options property could have fontSize, fontWeight, fontStyle, maxWidth, etc. and automatically shrink each line if it exceeds the max width (ImageSharp can do this too).