lukevp/ESC-POS-.NET #133: How to set an International Character Set like 12 "Latin America"

jupagose, 2021-09-29:
I'm trying to print spanish chars like á,é,í,ó,ú or ñ Ñ, 

I tryied with 
   
    e.CodePage(CodePage.PC852_LATIN2); // And others 

But my spanish charas still prints as extrange symbols.

There is a way to set the an International Character set for printing ?

djnuller, 2021-09-30:
If that doesn't work I would look at the code example I provided in the other issue.

jupagose, 2021-09-30:
> If that doesn't work I would look at the code example I provided in the other issue.

I made a mistake when assign the codepage, Just change
 
       _ = e.CodePage(codePage);  

with 
       
     lines.Add(e.CodePage(codePage));

This way the code page is included into the Byte[][] array. Another thing is, each line to print must be preceded with one code page.

I think codepage must be a printer setting and not an emitter setting, but works pretty well.

djnuller, 2021-09-30:
> I made a mistake when assign the codepage, Just change
>
> ```
>    _ = e.CodePage(codePage);
> ```
>
> with
>
> ```
>  lines.Add(e.CodePage(codePage));
> ```
>
> This way the code page is included into the Byte[][] [...]

Did you find a encoding that is working for you?

jupagose, 2021-09-30:
> Did you find a encoding that is working for you?

Yes, finally I opeted to use 
     
     CodePage.ISO8859_2_LATIN2

But some one other works too.

lukevp, 2021-10-01:
That's interesting, you should not need to set the code page except once when starting each print.  Did you try out and look at the CodePage tester in the ConsoleTest app?  It's pretty fully featured and lets you see all the characters for a given code page.  It only sets the code page one time and then prints every character in that code page, so I think that it's not necessary to set the code page repeatedly.

jupagose, 2021-10-01:
> That's interesting, you should not need to set the code page except once when starting each print. Did you try out and look at the CodePage tester in the ConsoleTest app? It's pretty fully featured [...]

I add each line into List<byte[]> then use printer.write method with  that list.ToArray(). This way codepage only works on first line

Y have nuget package 2.0 isntalled on my project.