lukevp/ESC-POS-.NET #137: Network printer no port specified

jupagose, 2021-10-04:
Hi,  I have a network printer OCPP-80582, this printer has no specific port for TCP connection, but when I try to connect to it, throw error asking me for port.

How can I do to connect without port? 

This is my code: 

     new NetworkPrinter(new NetworkPrinterSettings() { ConnectionString = $"192.168.1.144" })

I use this constructor due I can't use this on samples page:

    var printer = new NetworkPrinter(ipAddress: "192.168.1.50", port: 9000, reconnectOnTimeout: true);

Tks.

KevinAnass, 2022-03-09:
> Tks, I'm going to try with it.

i am using ethernet and i don't know the port of address i use the 80 but not work

arielbvargas, 2022-03-09:
You can try with port 6091.

KevinAnass, 2022-03-11:
> El mié, 9 de mar. de 2022 07:34, KevinAnass ***@***.***>

thank you so much the port by default is 9100 that is work for me