lukevp/ESC-POS-.NET #140: Stranger thing

hackdan, 2021-11-02:
The following error happens to me: it prints fine and after a while I sometimes printed strange characters ... watch the picture

https://ibb.co/SK1FrS3

igorocampos, 2021-11-12:
[image]

igorocampos, 2021-11-12:
Here you go, your image:

[image]

igorocampos, 2021-11-12:
Sounds like is something messing up with the image bytes buffer. See how it kind of started to print the same picture twice, without finishing it up first?

Do you have concurrent threads that send bytes to the same printer?

lukevp, 2021-11-13:
Kind of looks like the printer didn't receive the bytes correctly or overflowed the buffer.  I saw this type of printing happen a lot when I was printing the image.  Does the large images test pass successfully on this printer?  What about if you add 2x or 3x the # of images?

hackdan, 2021-11-23:
Ok, I'll try again with a simpler image and share the results with you.

hackdan, 2021-11-30:
Thank you by the support. Now it works like a charm.

closed 2021-11-30 by hackdan