lukevp/ESC-POS-.NET #146: Printer is hoding when printed one time, how to release the connection to let other user to print without closing the app/software?

hohinhenry, 2021-12-02:
Hi I am new to use ESC/POS.

I wrote a simple program to print with network printer, and I found that if user is not close the app/software after printed anything, the printer is not allow other connection until user close the app/software.

Is there anyway to release the printer connection without closing the app/software?

hohinhenry, 2021-12-02:
For example, I will use a button click to call below funcion:

**** Below is coding ******
```
private async void Print()
{
  string ip = "192.168.XXX.XXX";
  string networkport = "9100";
  var printer = new NetworkPrinter(settings: new NetworkPrinterSettings() { ConnectionString = $"{ip}:{networkport}" });
  var e = new EPSON();
  printer.Write(
  ByteSplicer.Combine(
    e.PrintLine("**TEST PRINT**"),
    e.PrintLine(""),
    e.PrintLine(""),
    e.PrintLine("")
    )
  );
}

```
=========
I can keep using this printer to print the same testing word or other things. But other user used the same app/software cannot access the printer until I close my app/software.

hohinhenry, 2021-12-02:
I also tried to use printer.Dispose() after the printing job is finished but looks like that is not the answer.

paul-turner-github, 2021-12-02:
I rolled back to 1.6 and it works.  Seems to be a 2.0 issue

hohinhenry, 2021-12-02:
> I rolled back to 1.6 and it works. Seems to be a 2.0 issue

Thanks!! That's work.

closed 2021-12-02 by hohinhenry

igorocampos, 2021-12-13:
@lukevp have you seen this issue? Perhaps you can easily identify what has changed about it?