lukevp/ESC-POS-.NET #147: Print width - Receipt Product and Prize on same line, with name left aligned and price right aligned

NewbridgeGareth, 2021-12-04:
Hi, 
First of all, this is a great library. Well done to all involved. 

I would like to know the best way of achieving the following if possible, preferably without using space characters. 

I have a product receipt and I want to left align the product name and right align the price on the same line. 
Is this possible?
I currently have a lovely looking receipt with the above format, however I have had to right align the price on the line below and it looks a little strange. 

I tried doing e.PrintLine(productname + e.RightAlign() + productprice),
but that didnt work and gave me a string output for the e.RightAlign(). 

Any guidance would be gratefully received. 

Thanks again

igorocampos, 2021-12-07:
Hello there, you will not be able to achieve that so easily. I believe the printer will override the alignment command when you send it twice in the same row. Maybe there's a printer model that behaves differently, but the ones I've worked with don't allow 2 alignments in the same line.

The solutions I've seen for this were the 2 you have mentioned:
1 - Create your own helper methods based on the printer's max char for one line and create the necessary space padding.
2 - Divide the text in 2 lines, one left aligned and the second right aligned.

nickcharlton, 2022-01-26:
I had to solve a similar problem recently, and you won't be able to do it without using spaces (not with a usual ESC/POS printer, anyway). This is the most elegant solution I came up with:

```csharp
private String EdgeAlign(string left, string right)
{
    var maxWidth = 48;
    var spaces = maxWidth - (left.Length + right.Length);

    return $"{left}{new String(' ', spaces)}{right}";
}
```

In case it's useful for anyone else coming across this problem!

elf-is, 2022-11-11:
Hi another easy way to do this is use C# string padding, you can do something like: `e.PrintLine(productname.PadRight(39) + productprice)` .
NB: You need to include the length of the string in the total number of characters you include in `PadRight`