lukevp/ESC-POS-.NET #155: Azure Web App won't let my "online" printer communicate.

Mafyou, 2022-01-02:
Hi,

Orignal Tweet: https://twitter.com/Mafyou/status/1476002420870234112

Clearly, in localhost, it works. But I don't want to use IIS.

Can someone, please, get my "outside" traffic in an Azure Web App. Why is it blocked? Insights?

Thanks a lot.

**To reproduce**

1. Create an ASP.NET DotNet Core PWA based on .Net 6
2. Copy/Paste on your Razor page the main ESCPOS_NET with a "NetworkPrinter".
3. Put the local IP:Port.
4. Try your local computer. It should work.
5. Find your public IP.
6. Forward the port on your router.
7. Try again with the public IP. It should work.
8. Publish it on an Azure WebApp. It doesn't work.

Why?

Regards,
Mafyou.

fatihbabacan92, 2022-01-28:
You simply can't print from a remote server. You need to run software within the local network of the printer.

Mafyou, 2022-01-28:
I just put a VPN and could be ok.
But anyway, I create a new thread, it just don't work, even locally.
Thanks for answering!

paul-turner-github, 2022-07-17:
I ended up creating a small windows service that connects to a Signalr endpoint.  When the print is triggered on the server, it sends a Signalr 'new job' and the service connects, grabs the resulting job and prints it locally.  It is pretty fast and responsive as well as being lightweight.