lukevp/ESC-POS-.NET #156: Printing long image just printed garbled message

hohinhenry, 2022-01-11:
Hi,

I found that printing long image with PrintImage function will print garbled message.

For example:
Print a image with 576 x 2517 is normal.
Print a image with 576 x 3027 is printing garbled message.

Also happening in ESCPOS_NET.ConsoleTest.

Thanks.

lukevp, 2022-02-04:
Can you try with the legacy print image too?  It is more compatible if you are not using an Epson printer. 

The garbled image issue typically happens when there is a buffer overflow on the printer side, and we haven't introduced a way to throttle the bytes written to the printer right now, so if your connection to the printer is so fast that it can overfill the buffer on writes, it will start dropping bytes out of the stream - causing the printer to interpret image data as print commands, which is why it goes all funky.

jjxtra, 2022-11-18:
Tried on TML-90 printer with 550x4096 image, got garbage. It did work with 550x2048 image.

jjxtra, 2022-11-18:
It's also worth noting that no images print whatsoever unless I specify legacy as true, maybe a limitation of my older printer model

jjxtra, 2022-11-18:
Made this method to break really large images into chunks, resolves the corruption issue even for the 550x4096 image:

```cs
    private static IEnumerable<Image<Rgba32>> ExtractChunks(Image<Rgba32> sourceImage)
    {
        const int chunkSize = 256;
        for (var y = 0; y < sourceImage.Height; y += chunkSize)
        {
            Rectangle sourceArea = new(0, y, sourceImage.Width, Math.Min(sourceImage.Height - y, chunkSize));
            if (sourceArea.Height < 1)
            {
                yield break;
            }
            Image<Rgba32> targetImage = new(sourceArea.Width, sourceArea.Height);
            int height = sourceArea.Height;
            sourceImage.ProcessPixelRows(targetImage, (sourceAccessor, targetAccessor) =>
            {
                for (int i = 0; i < height; i++)
                {
                    Span<Rgba32> sourceRow = sourceAccessor.GetRowSpan(sourceArea.Y + i);
                    Span<Rgba32> targetRow = targetAccessor.GetRowSpan(i);

                    sourceRow.Slice(sourceArea.X, sourceArea.Width).CopyTo(targetRow);
                }
            });
            yield return targetImage;
        }
```

nivle, 2023-06-14:
I ended up doing solving this like this, to solve some buffer issues, and some slow image printing issues.
I found out that if the image width is above a certain resolution(for my printer > 173), the printer will print the image very slow, 
kind of like a legacy printer, it will print the chunks created below one by one with a delay. Once I set the image resolution to 173(went -1 starting from 180(this is my printer dpi)), the image got printed much faster, not sure why this is the case.

```

IList<byte[]> data = new List<byte[]>
            {
                epson.CenterAlign()
            };

Image image = Image.FromStream("///");//add here your image

int size = 256;

for (int i = 0; i < Math.Round((double)image.Height / size, 0, MidpointRounding.ToPositiveInfinity); i++)
{
    int width = image.Width;
    int height = size * (i + 1) <= image.Height ? size : image.Height - (size * i);

    Bitmap newImage = new Bitmap(width, height);

    using (Graphics graphics = Graphics.FromImage(newImage))
    {
        graphics.DrawImage(image, new Rectangle(0, 0, width, height), new Rectangle(0, i * size, width, height), GraphicsUnit.Pixel);
    }

    using (MemoryStream memoryStream = new MemoryStream())
    {
        newImage.Save(memoryStream, ImageFormat.Tiff);
        data.Add(epson.PrintImage(memoryStream.ToArray(), false));
    }
}

data.Add(epson.FullCutAfterFeed(0));

return ByteSplicer.Combine(data.ToArray());

```