lukevp/ESC-POS-.NET #161: USB Connection error: "UnauthorizedAccessException" or "Access to the port 'COM3' is denied."

shajeermhmmd, 2022-01-30:
USB com port is connected and printing successfully, but while trying to print again I am getting the issue with "UnauthorizedAccessException" or "Access to the port 'COM3' is denied." with the com port.
How can I solve this issue?

printer = new SerialPrinter(portName: "COM3", baudRate: 115200);

igorocampos, 2022-02-03:
Are you using a physical COM port, or an adapter/emulator?

shajeermhmmd, 2022-03-17:
I have two printers One is Epson's official printer and another one is a local company made, it is working perfectly with Epson's printer, but not with another one.

is there any compatible way to expose the USB interface as a virtual Serial port in windows?

[is there any compatible way to expose the USB interface as a virtual Serial port?](https://github.com/lukevp/ESC-POS-.NET/issues/167)