lukevp/ESC-POS-.NET #162: Reconnection delay or interval to next print.

elle-pinto, 2022-01-31:
I am using network printer to print. First one successfully prints. When I try to print once again it does not come. After few minutes when it is requested again it prints. There is an unknown delay or interval to print.

elle-pinto, 2022-01-31:
Could you please tell me how can I solve this issue

shajeermhmmd, 2022-01-31:
@lukevp  I am also facing the same issue... anybody has a solution for this...?
Bellow is the code used to connect with Network.

var networkSetting = new NetworkPrinterSettings();
networkSetting.ConnectionString = "192.168.100.7" + ":" + "9100";
printer = new NetworkPrinter(networkSetting);

The first print will work perfectly, but I have to wait for 1-3 minutes to get another print... 

while debugging I am getting the following exceptional error on the second print.

'((System.Collections.ICollection)((ESCPOS_NET.BasePrinter)printer).WriteBuffer).SyncRoot' threw an exception of type 'System.NotSupportedException'

danielgey, 2022-02-02:
Same problem solved downgrading to version 1.6

elle-pinto, 2022-02-03:
> Same problem solved downgrading to version 1.6

Thank you, this solved my problem.

lukevp, 2022-02-04:
Sorry for the delay in response @elle-pinto @danielgey  @shajeermhmmd . Which platform/.net version were you all running this on?  V 2.0 introduced persistent connections to printers and it manages reconnecting transparently.  Sounds like some of the logic isn't working for the platform you are using and causing these delays.  1.6 didn't properly manage long lived connections, and it often failed to reconnect (especially when changing paper or other actions that cause the printer to go offline and back online quickly).  Sounds like there is still some more work to do on 2.0.  Reopening this issue to track.

danielgey, 2022-02-04:
> 
Hi @lukevp, 
My app is running on .NET 6.0

elle-pinto, 2022-02-05:
Hello @lukevp My .Net application is running on v5.0. But still faced this issue of reconnecting.

shajeermhmmd, 2022-02-05:
@lukevp Thanks for reopening the issue, I am also using the .net version V5.0 and the plugin version 2.0, as I mentioned above the error comes while debugging... and it will print after 2 or 3 minutes without doing anything... and solved by downgrading the Plugin... now I can print continuously but I am expecting the issues which you mentioned in the comment with the version 1.6

var networkSetting = new NetworkPrinterSettings();
networkSetting.ConnectionString = "192.168.100.7" + ":" + "9100";
printer = new NetworkPrinter(networkSetting);

The above lines of code I am calling in each button click to print. Is there any other way to initialize the object for the printer? or it will be good if you can share some code that is successfully running so that I can verify the logic which I used for the same.

Thank you

elle-pinto, 2022-02-10:
@lukevp is there any update on this plugin. Please let us know when this is resolved.

Solvengo, 2022-02-14:
@lukevp in 2.0, The printer automatically reconnects (or updates the printer status, we only print when the online status of the monitoring is true) when opening and closing the tray of the printer. At least it does that with the wifi printer.

We used to dispose the printer when there were no commands, and reconnect when there were new print commands. But I suppose that is not needed anymore?

fatihbabacan92, 2022-02-14:
@lukevp in 2.0, The printer automatically reconnects (or updates the printer status, we only print when the online status of the monitoring is true) when opening and closing the tray of the printer. At least it does that with the wifi printer.

We used to dispose the printer when there were no commands, and reconnect when there were new print commands. But I suppose that is not needed anymore?

rajndrasharma, 2022-05-30:
But when we are connecting from two different places, then problem still persist. If i gives print command from one pc it is perfect but when i give print from 2nd PC it doesnt print.