lukevp/ESC-POS-.NET #168: Korean characters are being printed when Arabic CodePages are specified

lerasah, 2022-03-22:
Printer: Yujin Thermal Printer
Windows

Tests done: Printed arabic using Notepad / wordpad/ msword, prints perfect. 
Printed Arabic characters without codepage, prints question marks.

Thanks.

Code:

              var e = new EPSON();
                Encoding.RegisterProvider(CodePagesEncodingProvider.Instance);
                Encoding enc = Encoding.GetEncoding(1256);

                var bytes = ByteSplicer.Combine(
                        e.CodePage(CodePage.WPC1256_ARABIC),
                        e.CenterAlign(),
                        e.PrintLine(" Test Page See the Arabic text Below WPC1256_ARABIC"),
                        enc.GetBytes("طباعة صفحة إختبار "),
                        e.PrintLine("\x1D\x56\x42\x00"),
                        e.CodePage(CodePage.WPC1256_ARABIC),
                        e.CenterAlign(),
                        e.PrintLine(" Test Page See the Arabic text Below WPC1256_ARABIC"),
                        enc.GetBytes("طباعة صفحة إختبار "),
                        e.PrintLine("\x1D\x56\x42\x00"),
                        e.CodePage(CodePage.PC720_ARABIC),
                        e.CenterAlign(),
                        e.PrintLine(" Test Page See the Arabic text Below PC720_ARABIC"),
                        enc.GetBytes("طباعة صفحة إختبار "),
                        e.PrintLine("\x1D\x56\x42\x00"),
                        e.CodePage(CodePage.PC720_ARABIC),
                        e.CenterAlign(),
                        e.PrintLine(" Test Page See the Arabic text Below PC720_ARABIC"),
                        enc.GetBytes("طباعة صفحة إختبار "),
                        e.PrintLine("\x1D\x56\x42\x00"),
                        e.CodePage(CodePage.PC864_ARABIC),
                        e.CenterAlign(),
                        e.PrintLine(" Test Page See the Arabic text Below PC864_ARABIC"),
                        enc.GetBytes("طباعة صفحة إختبار "),
                        e.PrintLine("\x1D\x56\x42\x00")
                );`

(ignore the paper cuts)
[image]

AmerAlmadanydotNet, 2022-05-11:
same problem ....

savindu-pasintha, 2022-08-31:
you cant arabic text draw in canvas object and then you can pass canvas bitmap object to printer. it will be print as a image.

lerasah, 2022-09-02:
> you cant arabic text draw in canvas object and then you can pass canvas bitmap object to printer. it will be print as a image.

Hey Savindu bro - thanks for the reply. I was able to print arabic using a similar gimmick for now, but there are issues like font anti aliasing is not smooth. That's why I wanted to find the issue with the library method. `SendBytesToPrinter`

savindu-pasintha, 2022-09-03:
also me , faced that issue.

savindu-pasintha, 2022-09-03:
[image]
see, image is not smoot, you are correct. these days I am also trying to fix that one.

mo5br, 2025-05-10:
> [image] see, image is not smoot, you are correct. these days I am also trying to fix that one.

did you find a solution?

Kremed, 2025-10-22:
any one find a solution for this issue ??