lukevp/ESC-POS-.NET #169: 2.0.0 does not disconnect on disposing NetworkPrinter

AertsIT, 2022-03-24:
When disposing the NetworkPrinter, it looks like the connection is not terminated.

gusVLZ, 2022-03-31:
I had the same issue, the application printed once and stopped working, the solution was to implement a Singleton architecture to maintain the same connection with the printer, it solved for me

orbitasol, 2022-06-22:
Hi @gusVLZ I am facing the exact same problem as you and would like to know if you can share a sample code of your solution.  Please note that I am using the library in an Asp.net Core 5 MVC project. Thanks in advance.

lukevp, 2022-08-18:
Give the new ImmediateNetworkPrinter a try in release 3.0.0 @AertsIT @orbitasol , it does not have these issues with repeated disposals.  The NetworkPrinter implementation in 2.0 was meant to be a singleton for the lifetime of your application, as it handles reconnecting internally, which is what Gus recommended.  The ImmediateNetworkPrinter is simpler to use, but it loses the status functionality that the singleton approach has.

closed 2022-08-18 by lukevp