lukevp/ESC-POS-.NET #170: Flush buffer?

pm64, 2022-03-31:
I managed to get this library partially working with an Epson TM-m30II.

The only issue is that the printer doesn't print immediately after receiving the data. Rather, the printing happens several minutes after the data is received, perhaps after I've sent more data, or some period of time has passed.

My first thought was that some buffer needs to be flushed after the data is sent -- but that idea is not discussed anywhere in this repository. What other explanation could there be for this behavior?

igorocampos, 2022-05-11:
If you provide more details and debug testing so we know which of the described behaviors are happening would be much better.
For example:
- What are you trying to print? Simples text or barcodes?
- Which one is true: it prints after you send the next data or only when you wait X amount of time?
- When you debug the code, where the execution sits while the printer hasn't printed anything? Is it in a loop of any kind?

rajndrasharma, 2022-05-30:
I am generating receipt in loop for different group. Now if printer is on same address, it is not printing immediately. it takes some time to print the same and in debugging mode when i stop debug it will print remaining receipts. Kindly can you please provide solution for the same.

lukevp, 2022-08-18:
@rajndrasharma @pm64  try using the new ImmediateNetworkPrinter that will be available in 3.0.0. #189  introduced this new implementation.  It sounds like you are both using the network printer, and the other network printer implementation is intended to be used by creating a single instance and keeping a handle to it for the lifetime of your application. If you don't use it that way, it gets kind of flaky in the reconnect logic.  The ImmediateNetworkPrinter directly connects and writes the byte stream, giving you control over awaiting it and handling connection timeouts and retries yourself. I'm closing this issue since that is likely the cause of your issue.

closed 2022-08-18 by lukevp