lukevp/ESC-POS-.NET #173: Little help with epson serial printing, nothing comes out

Substractive, 2022-04-24:
I've tried ESC_POS_USB_NET and it works great. My epson TM-T20III is configured to work on COM4 and test print works fine.
Example from readme does not work at all.
Here are screenshots of printer config. Any idea where to look?

[image]
[image]
[image]

igorocampos, 2022-05-11:
if you go to the command line and send some text to the COM port, does that work?
Something like
```
C:\ DIR > COM4
```

us259er, 2024-08-31:
Hi. I am experiencing the same issue, also on TM-T20III but on COM5 port. When I do:
`echo abcd > \\.\COM5`
My printer prints the text, I have also configured it to work with the python-escpos library and there it works just fine