lukevp/ESC-POS-.NET #174: TCP Connections to Network Printer from multiple clients not working

NewbridgeGareth, 2022-05-06:
Hi

Forgive me if I have overlooked something here, but my understanding (which may well end up being incorrect) is that a server (or in this case, a network printer) supports multiple connections on the same port (9100) through the concept of the tuple. 
(SourceIP:SourcePort|HostIP:HostPort|Protocol). 

I have successfully integrated the ESC-POS packages in to my application and it prints beautifully over TCP and all the status feedback works great. Tested it in various scenarios such as powering off, opening lid etc. 

However....if I try to make a connection to the printer using the same IP and Port as an application running on a different machine is using and connected to, the connection is refused. 
"No connection could be made because the target refused it". 

If I run Netstat on the second machine, I can see that the an attempt to connect to the IP and port was sent and it waiting. 
If I close the running application that has the active connection, the second connection succeeds. 

Based on this, I can only draw the conclusion that either something in the ESC-POS library is blocking multiple connections, or the printer device (Aures OPD333) simply doesn't support multiple connections, which I find strange. 

Any help that anyone could give me would be greatly appreciated. 

TIA

igorocampos, 2022-05-11:
From what you described, sounds like the connection is kept open by your first application, even when no data is being sent to the printer. 
I don't recall if ESC-POS library has a connect/disconnect method for TCP, but if not you can always dispose the instantiated printer class that connected and it should be enough to free the address for a new connection. In other words, Connect, Print and Disconnect, every time you wish to use the printer.

Other than this, you would have to dive into the library code and propose a working alternative through a PR.

NewbridgeGareth, 2022-05-13:
@igorocampos Thank you so much for taking the time to comment. I had actually decided already to adopt the Connect/Print/Disconnect approach and I have it working nicely. It's not something I can contribute as a PR though unfortunately as it is too far removed from the ESC-POS core functionality now. Thanks again.

closed 2022-05-13 by NewbridgeGareth

lukevp, 2022-07-08:
As a follow-up, it does appear that some non-Epson printers are implemented where they don't support multiple inbound connections.  The ESCPOS library is only tested with Epson printers (since that's what I have) as well as a Royal PT-300 for wifi/bluetooth tests.  I think this makes a strong case for having 2 different implementations, one that keeps a persistent connection (for printers that support multiple connections, or use cases where the POS device can claim exclusive ownership), and don't cause other applications to malfunction) and another for fire-and-forget printing that doesn't maintain a connection long-term.

NewbridgeGareth, 2022-08-09:
Hi Luke,

Hope you’re well mate.

Do you know of a way to transpose the byte array that is sent to the printer in to either of the following:

  *   PrintDocument
  *   PDF
  *   XPS

The idea being to be able to bypass the array going to a Thermal printer but instead display the output, including all formatting and any images/barcodes etc, on the screen. Like print preview.

Regards,

***@***.***
Newbridge is a trading name of Newbridge Software Ltd Company Number (10510559)
This e-mail is from Newbridge Software Ltd It is confidential to the addressee and may contain copyright and/or legally privileged information. No-one else, other than on behalf of the addressee, may read, print, store, copy, forward or act in reliance on all or any of it or its attachments. We accept no liability for any loss, damage or consequence resulting either directly or indirectly from the use of the attached files. If you receive this e-mail in error, please telephone us on 029 20 990810.
***@***.***<https://www.facebook.com/newbridgesoftware/>       ***@***.*** <https://www.instagram.com/newbridgesoftware/>        ***@***.*** <https://www.linkedin.com/company/newbridge-software/>

lukevp, 2022-08-11:
@NewbridgeGareth  I'm not aware of any ESCPOS parser that can output PDF or any other format.  I've been interested in building something like this (to provide print previews, etc.) Can you open up a separate issue that we could discuss about this and maybe document any links you find to anything existing that does this?  This issue isn't really related to this topic.