lukevp/ESC-POS-.NET #177: Nothing happens with UWP

Letippi, 2022-05-30:
Hi, I have tried with both versions 1.6.0 and 2.0 but nothing happens on UWP. The same code with Android, IOS and MacOS works perfectly.

With version 1.6.0 the code is:
 ep = new EPSON();
 String ip = "192.168.0.188";
 int networkPort = 9100;
 printer = new NetworkPrinter(ipAddress: ip, port: networkPort, reconnectOnTimeout: false );
printer.Write(ep.PrintLine("HELLO WORLD"));

With version 2.0 the code is:
 ep = new EPSON();
 String ip = "192.168.0.188";
String networkPort = "9100";
string ip_port = $"{ip}:{networkPort}";
NetworkPrinterSettings network_parameters = new NetworkPrinterSettings() { ConnectionString = ip_port };
 printer = new NetworkPrinter(settings: network_parameters);
 printer.Write(ep.PrintLine("HELLO WORLD"));

Anyone help?

Thankyou

lukevp, 2022-08-03:
Does UWP need permissions at the application level to make an outbound network connection? @Letippi

jupagose, 2024-03-04:
UWP reports Acces deniied to COM port, I use same base code to print in maui app and it prints ok. I added point of service, Low level and Low level devices but it does not works.

jupagose, 2024-03-04:
After many tryies Finally I take totally new one UWP project and tryed  opening an standard serialport an print a line 

`        private SerialPort _serialPort { get; set; }
        private void Page_Loaded(object sender, RoutedEventArgs e)
        {
            _serialPort = new SerialPort("COM3");
            _serialPort.Open();
        }

        private void Button_Click(object sender, RoutedEventArgs e)
        {
            _serialPort.WriteLine("Hola mundo puerto com3");
        }
    }
`

Ths code works fine but If  I install ESCPOS_NET the problem starts "Acces denied to port com"