lukevp/ESC-POS-.NET #178: Network printer block further connection in 2.0.0

rajndrasharma, 2022-05-30:
I was using ver 2.0.0 for print of receipt, it was working fine if I keep conection alive. If I dispose printer, it wont get connected again. So I made one single connection and pass it around to print. Now connection got refused when i try to connect from another PC. 

Downgrading to 1.6.0 solved this issue.

So i think in ver 2.0.0 printer is not getting disposed or disconnected correctly.

SORencber, 2022-06-01:
I have same Problem.
Version 2.00 is not working.

lukevp, 2022-08-18:
Try out version 3.0 which has a new network printer implementation called ImmediateNetworkPrinter. It solves the issues of disposing and reconnecting in 2.0.  The network printer in 2.0 was intended to be used as a singleton, but that was not clear to many people, so this new implementation works more how you're thinking, where you can dispose of it anytime and you can new them up whenever you want. It also doesn't buffer anything so you can directly await and handle exceptions.

closed 2022-08-18 by lukevp