lukevp/ESC-POS-.NET #186: USB support

rajndrasharma, 2022-07-29:
Can anyone please suggest how to print on thermal printer with usb?

arielbvargas, 2022-07-29:
Unfortunately, I wasn´t able to make a thermal print with USB on Windows. Network printing works great. If anyone can share working code, I will appreciate it. Thanks.

lukevp, 2022-07-29:
Hi @rajndrasharma @arielbvargas , did you follow the directions in the Readme to install the drivers? USB printing works fine on windows, Mac, and Linux, but the way it works is through virtual serial port emulation. It doesn’t work with native USB devices. Let me know what printer model you have and what virtual port you’ve tried!

rajndrasharma, 2022-07-29:
It is rugtek and tvs rp 3200 series printers. I am not able to add virtual com ports. I will try once again.

hackdan, 2022-07-29:
Anyway, the TM EPSON on TM Virtual Port always keep emulating as the printer was connected even if the cable is disconnected

lukevp, 2022-08-03:
You should be able to use the status features to determine if the printer is actually connected and online, independent on if the COM port is connectable @hackdan .

lukevp, 2022-12-24:
I changed this to bug status, we need to research if the disconnected state can be detected and reflected using the virtual port.  Most likely if you disconnect the printer you would have to just have a timeout on the write or something to that effect.  Even if you use automatic status, that is depending on the printer communicating its status.  The latest version of the lib has a status poller, you could try that and see if you can detect when the printer is offline.

DRaichle, 2023-04-26:
I have the same problem with an Epson TM-T20-III and I just debugged it a little bit. If the printer is disconnected/switched off the member "DsrHolding" of the serial port changes from true to false. After connecting/switching on it changes to true again.
Additionally there is a PinChanged-event in the SerialPort with an event-arg of enum-type "SerialPinChange" that contains a value named "DsrChanged". => This might be used for setting the connected-status and to invoke the connected-event.

The meaing of DSR according to wikipedia is: Data Set Ready which means that the DCE (in this case the printer) is ready to receive and send data.

DRaichle, 2023-04-26:
I have just seen that it is possible to define in the Epson Virtual Port Assignment Tool how the signal lines (DSR, CTS, RI, DCD) shall be used. Standard setting is: DSR and CTS are set if printer is online - one alternative setting is to set a line if "Printer Power is ON"