lukevp/ESC-POS-.NET #201: Random timeout issues with large job volume

SILENTqt, 2022-10-24:
While this may not be related to this specific release, someone may have encountered this previously - 
Sits with this style of error for 10 minutes or so...
[image]

jjxtra, 2022-12-24:
Which printer class are you using?

SILENTqt, 2023-02-08:
> Which printer class are you using?

Epson TM82III

jjxtra, 2023-02-08:
I don't know if you are chunking your images into heights of no more than 2048 (1024 ideally) but that might be a good start

SILENTqt, 2023-02-08:
We only include one image and its much smaller then 1024

SILENTqt, 2023-02-08:
ImmediateNetworkPrinter

SILENTqt, 2023-02-08:
Is it best to call printer.WriteAsync in chunks of max 1024 bytes