lukevp/ESC-POS-.NET #203: C# Api implementation error

maximilianoCalderon, 2022-11-11:
Hi Team, first of all i want to thank for this great and useful work!
I have an API implementation for a POS Software (C#, .NET 5.0), i was using the 1.6.0 version and was running perfectly. This is my code

koo9, 2022-11-11:
I think the issue that im experiencing is the same as your. see the description here. 
https://github.com/lukevp/ESC-POS-.NET/issues/194

maximilianoCalderon, 2022-11-14:
@koo9 Yes it is, i have the same issue than you, the only way to print in 3.0 its debugging and adding a breakpoint.

orlandommb, 2023-01-04:
> > @koo9 Yes it is, i have the same issue than you, the only way to print in 3.0 its debugging and adding a breakpoint.
> 
> yes breakpoint will give it enough time to connect to the printer.

Not an expert or have any experience using this nuget, came here to check it out for a project, i was reading the documentation and saw that it has support for inmediate network printing and you need to use await printer.WriteAsync, maybe that gives you time to connect to the printer.