lukevp/ESC-POS-.NET #210: Find the number of characters in a string

Kamaew, 2022-12-01:
Hello

Can anyone suggest how to find out the number of characters that fit in one line?
For example, to make a separator line
------------------------
and she was always from edge to edge.

Many Thanks

igorocampos, 2022-12-02:
That varies from printer to printer. I think I have a suggestion open here to implement this as a property in the class and you could inform the columns of your printer when creating a new instance of the class. That would allow having commands such as the one you just described, a full line print.

However, that information can't be obtained by this library, you are the one that needs to inform it.

Kamaew, 2022-12-05:
That would be good