lukevp/ESC-POS-.NET #212: More flexible font size

Kamaew, 2022-12-01:
Hello

Can someone tell me how to fine tune the font size? double width and height doesn't give any flexibility. In fact, the printer supports setting sizes for almost every character. How to implement it in the library?

Many Thanks

igorocampos, 2022-12-02:
I'm only aware of 2 ways of changing fonts in ESC/POS:
1 - The mentioned width and height multiplier (which can very well be more than the double - e.g. 4x, 8x, etc)
2 - The secondary font type (there's a font A and B you can use) but that's more of style than size.

Other than that you would need to research ESC/POS commands that can change it, and if you find it, it should be very easy to implement the commands in the library.

lukevp, 2023-06-17:
Closed for staleness and marking for possible documentation updates.

closed 2023-06-17 by lukevp