lukevp/ESC-POS-.NET #213: Printer Prints Twice

ChaseDDevelopment, 2022-12-07:
Hey there! I am running this package on .NET 6.0, on Ubuntu 22.04. 

I initialize the printer object like so:

```private FilePrinter printerObj = new FilePrinter(filePath: "/dev/usb/lp5");```

followed by:

```var e = new EPSON();```

and Essentially the Example receipt from the readme. 

My issue comes from the fact that sometimes the printer will print 2 copies of the receipt back to back from one button_click. 

I have tried Flushing the printer immediately after the print command like so:

```            
            printerObj.Write(GetReceipt());
            printerObj.Flush(printerObj, null);
            System.Threading.Thread.Sleep(5000);
```

Perhaps I am flushing it wrong (Do I need to pass in different parameters)? or if anyone knows why the receipt will print twice randomly. I am using an EPSON TM_T20III

lukevp, 2022-12-23:
@ChaseDDevelopment Nothing about this library would cause it to print twice, and it's not necessary to manually flush the printer object.  It autoflushes by default.  Are you sure that the button click event isn't firing twice? That would be the most likely cause of this issue. Do you have a minimal reproducible version of this?  I have a TM_T20II and I haven't ever seen a double print.

lukevp, 2023-01-27:
Closed stale issue.  Ping me if you get a chance to try this again.

closed 2023-01-27 by lukevp

luizfern12, 2023-12-16:
I have the same problem and i know for sure the print isn't being called twice by the frontend

butskristof, 2025-02-10:
Did anyone ever figure this out? I'm running into the same issue using FilePrinter and a TM_T20II connected over USB. 
My application is running in Docker on a Raspberry Pi, but the same happens in a Linux VM on my (Apple Silicon) Mac.