lukevp/ESC-POS-.NET #218: Baseprinter Dispose() method block printing

emanuhell, 2023-01-23:
Baseprinter Dispose() method block printing. For example:

```
printer.Write(Tests.Receipt(e));
printer.Dispose();
```

prints nothing!
I need the Dispose method to release the connection to the printer after each print, to allow other running applications to use the same printer.
Thanks.

lukevp, 2023-01-27:
Which printer are you using?  Some of the implementations are asynchronous (Writes are buffered and flushed by a task).  This is to increase the performance of many small writes.  You may just be disposing the connection before the flush happens.  We should probably expose a way to await the print queue being empty.

emanuhell, 2023-01-30:
`Which printer are you using?`
I'm using the SerialPrinter.

`We should probably expose a way to await the print queue being empty.`
What method are you talking about? I can't find it.

Thanks.

xZeroKooLx, 2023-02-02:
I have a similar issue with trying to dispose of a networked Epson . If something  else tries to print to the Pinter after my connection is open then it fails and so do the rest of my prints.

 private void PrintNumbers()
        {
            string filePath = "textfile.txt";
            string textFromFile = File.ReadAllText(filePath);
            string[] parts = textFromFile.Split(':');
            string targetAddress = parts[0];
            string targetPort = parts[1];
            bool monitor = true;
            if (printer == null) printer = new NetworkPrinter(settings: new NetworkPrinterSettings() { ConnectionString = $"{targetAddress}:{targetPort}" });
            //printer = new NetworkPrinter(new NetworkPrinterSettings() { ConnectionString = "192.168.1.129:9100" });
            Bitmap bmp = new Bitmap(listBox1.Width, listBox1.Height);
            listBox1.DrawToBitmap(bmp, new Rectangle(0, 0, bmp.Width, bmp.Height));

            e = new EPSON();
            // Convert the bitmap to a byte array
            using (var ms = new MemoryStream())
            {
                bmp.Save(ms, ImageFormat.Png);
                byte[] img = ms.ToArray();
                // Print the image
                Setup(monitor);
                //printer?.Write(e.FeedLines(1));
                printer?.Write(e.CenterAlign());
                printer?.Write(e.PrintLine(customMsgRTextBox.Text));
                printer?.Write(e.FeedLines(1));
                printer?.Write(e.SetStyles(PrintStyle.Bold | PrintStyle.FontB | PrintStyle.DoubleHeight | PrintStyle.DoubleWidth));
                printer?.Write(e.PrintLine("YOUR LUCKY NUMBERS ARE:"));
                printer?.Write(e.FeedLines(1));
                printer?.Write(e.PrintImage(img, true, true, 350, 1100));
                printer?.Write(e.SetStyles(PrintStyle.None));
                printer?.Write(e.FeedLines(1));
                printer?.Write(e.PrintLine("Dont forget to put your name on the back of"));
                printer?.Write(e.PrintLine("your ticket and place it in the red bin located"));
                printer?.Write(e.PrintLine("in the bar for your chance to win 1 of 2x $100"));
                printer?.Write(e.PrintLine("Telegraph vouvhers in our"));
                printer?.Write(e.PrintLine("end of month draw!"));
                printer?.Write(e.FeedLines(1));
                printer?.Write(e.PrintLine("Thanks for visiting The Telegraph Gunning."));
                printer?.Write(e.PrintLine(DateTime.Now.ToString("dd/MM/yy hh:mm:ss tt")));
                printer?.Write(e.FeedLines(3));
                printer?.Write(e.FullCut());
                    printer.Dispose();
            }
        }