lukevp/ESC-POS-.NET #219: USB printing on macOS

butskristof, 2023-01-26:
Hi! I'm trying to use this library to print receipts from a POS application. However, I'm facing some trouble getting my printer to show up as an available device. 
For development, I'm using a MacBook Pro with M1 Pro chip (so arm, not x86). Eventually, the software should run on a Raspberry Pi so I'm taking a Linux approach overall. The printer is an Epson TM-T20II with USB and serial ports, connected via USB directly to my laptop. 

First, I had the printer show up as a 'normal' printer in my system preferences, and I could select it as a print target in e.g. my text processor. Since this doesn't seem like "raw" printing mode, I've used the TM utility to change from "Printer Class" to "Vendor Class". This made the printer issue a short buzz, and since then it's listed as "offline" in my printer menu. 

Going by the USB usage guide in the readme, I should see the printer exposed as a file under `/dev/ttyusb` or a variation. Unfortunately, neither of them seem to exist on my machine after connecting the printer. Some files I've looked for: `/dev/ttyusb`, `/dev/ttyusbserial`, `/dev/usb/lp*`, `/dev/tty.*`, ... 
In my application, running `SerialPort.GetPortNames()` only yields `/dev/tty.Bluetooth-Incoming-Port` and `/dev/cu.Bluetooth-Incoming-Port`. 
Going by some of the existing issues, the "virtual COM port driver" seems to come up now and again, but that's probably not applicable here? 

Am I missing something here? Is USB printing on macOS possible at all? Would using a parallel to USB cable instead work? 
I'm willing to explore some options here, but am a bit clueless where to start looking. 

Thanks in advance!

lukevp, 2023-01-27:
Hi @butskristof ,

Yes, USB printing on macOS should be possible. It was only tested in 1.x releases, but the FilePrinter implementation hasn't changed since then.  I haven't tested it on an M1, but I don't think that should make any difference, and it seems like your problem is related to getting the printer available in the first place.

The virtual COM port driver is only really useful on Windows, because Windows doesn't expose a native serial-level communication for USB devices unless a driver implements it.  On Linux and MacOS, the lower level direct serial communication to the printers is exposed as a file handle, much like many other devices on those platforms.  That's the thing that shows up in /dev/ttyusb*.  I don't think having a higher-level print drive for not having it would make any difference as far as ability to print using the file handle.

I have the same MacBook as you, as well as a TM-T20II USB - funny coincidence!  I'll give it try this weekend and see if I can reproduce your issue or point you in the direction of how this specific printer mounts on MacOS.

butskristof, 2023-01-30:
Thanks for the clarification @lukevp ! At least now I know I was on the right path :-). 
 
Just to verify, I've loaded up Linux Mint on another computer and connected the printer. There actually was a `/dev/usb/lp0` file handle there. Could it be that I'm just looking over the mount point in macOS? There were a lot of tty devices, just none that pointed to USB or LP. 
Is there any command I could use to single out one of these as the printer?

lukevp, 2023-02-02:
@butskristof looked into this a little more.  One easy way to tell if the device is mounting is just to capture the device list to a file, then plug it in, and see if there are any new devices.  something like this:
`ls -al /dev > ~/devices-before.txt`
plug device in
`ls -al /dev > ~/devices-after.txt`
then you can use `diff` cli command or do what I do - open one file in vs code and do a `Compare Active File with...` command to compare to the second file.

My findings so far: the device did not mount for me either on macOS 13.1 with an M1 Pro.  I looked into this a bit and it seems that M1 compatibility is lower with receipt/label printers out of the box, so we will likely need a custom driver.  I looked on Epson' site and found a download for the TM series drivers for Mac.  the problem is that it only lists up to macOS 12, and only lists the TM-T20III.  
[image]

The good news is that the M1 is listed there for the newer OSes.  It's possible that the driver is still compatible with macOS 13 and with the TM-T20II (and it's just not listing those 2 because in the case of the OS it was released before 13 came out, and in the case of the printer, they only list the latest model of that line). However, even installing the driver doesn't expose the device handle anymore and only supports USB-level printing.  That would be a bummer and a pretty major functional regression but it's definitely possible that that's just how the driver works now.

These are my suggested options for you:
- use RS232 and a generic usb/serial interface that should happily mount to /dev/, probably with no drivers.
- switch to a wifi or Ethernet printer (these both definitely work from Mac as this interface is not dependent on the host at all)
- reach out to Epson to inquire about using a /dev/ mount with the TM-T20II on MacOS 13.1.  If you do this, feel free to CC me as well, as I'd be very interested to hear from them about this support.

lukevp, 2023-02-02:
After re-reading your original post, your final destination is a Raspberry Pi anyway.  Might I suggest you just connect the printer directly to your Raspberry Pi?  You can then use VS Code Remote Development to work on the scripts from your Mac but have a remote terminal on the Raspberry Pi for testing and issuing commands.

butskristof, 2023-02-03:
Hi Luke, thanks for the detailed research! 

Bummer that the Apple Silicon support isn't there and that we're dependent on Epson driver support now. It's only to be expected Epson will stop providing these drivers now. 

As I mentioned before, on another computer with a Linux installation the device seemed to show up so I'll probably just use that for developing the part of the application that actually requires the printer (and the final deployment). 
Good to know that a generic RS232/USB would probably work as well on my Mac, should I ever have to go that route. 

Thanks again for your help!