lukevp/ESC-POS-.NET #222: Dispose not working correctly

xZeroKooLx, 2023-02-08:
Im using BasePrinter to print to a Epson epson tm-t82ii over a networked connection. The issue I'm facing though is that printer.Dispose(); does not seem to be full disposing of the connection. once i print using the following code no other applications or devices can send data to the printer until i close my application.

`private void PrintNumbers()
{
bool monitor = true;
if (printer == null) printer = new NetworkPrinter(settings: new NetworkPrinterSettings() { ConnectionString = $"{targetAddress}:{targetPort}" });

        e = new EPSON();
        // Convert the bitmap to a byte array
        using (var ms = new MemoryStream())
        {
            bmp.Save(ms, ImageFormat.Png);
            byte[] img = ms.ToArray();
            // Print the image
            Setup(monitor);
            //printer?.Write(e.FeedLines(1));
            printer?.Write(e.CenterAlign());
            printer?.Write(e.PrintLine(customMsgRTextBox.Text));
            printer?.Write(e.FeedLines(1));
            printer?.Write(e.SetStyles(PrintStyle.Bold | PrintStyle.FontB | PrintStyle.DoubleHeight | PrintStyle.DoubleWidth));
            printer?.Write(e.PrintLine("YOUR LUCKY NUMBERS ARE:"));
            printer?.Write(e.FeedLines(1));
            printer?.Write(e.PrintImage(img, true, true, 350, 1100));
            printer?.Write(e.SetStyles(PrintStyle.None));
            printer?.Write(e.FeedLines(1))
            printer?.Write(e.FeedLines(1));
            printer?.Write(e.PrintLine("Thanks for visiting The Telegraph Gunning."));
            printer?.Write(e.PrintLine(DateTime.Now.ToString("dd/MM/yy hh:mm:ss tt")));
            printer?.Write(e.FeedLines(3));
            printer?.Write(e.FullCut());
            printer?.Dispose();
        }
    }`

lukevp, 2023-02-25:
Have you tried using a using block to dispose of the printer instead of calling Dispose yourself?  Could you put together a minimal reproduction that causes the issue?

lukevp, 2023-06-17:
Closing for staleness, let me know if you have a chance to reproduce this and we can reopen the issue.  I haven't seen this issue before, so I don't have a reproducible example to investigate.

closed 2023-06-17 by lukevp