lukevp/ESC-POS-.NET #223: Printing support using Windows spooler

sinni800, 2023-02-14:
As USB printing is currently not possible and some weird china printers don't seem to really have a good way to make a virtual com port (at least I havent found any) I think a good way to make this possible is to have a "Windows Printer Spooler" printer backend.

This project here (same license, MIT) has such an implementation: https://github.com/mtmsuhail/ESC-POS-USB-NET/blob/master/ESC-POS-USB-NET/Helper/RawPrinterHelper.cs

Essentially, creating a RAW printer in Windows and sending ESCPOS to it like you would to a COM port works just fine like that, but this other library is not nearly as good as this one...

lukevp, 2023-02-25:
I'm super interested in enabling native USB printer through the windows spooler!  The only criteria I would have are:

1) library still works on all platforms (it is currently compatible with .net standard and runs on Windows, MacOS, Linux, Xamarin iOS, Xamarin Android).   
2) it uses raw printing still even though it's going through the windows spooler
3) the usage and setup can be documented in the README.

Is this something you'd be interested in contributing?  I think it would be really valuable for a lot of people who can't use the virtual COM, it's just not something I was familiar with on how to implement.

sinni800, 2023-02-25:
Hmm, I have hacked together something that works perfectly fine for my things, but I haven't put any junctions in that would properly make it fail on Non-Windows.

Since I have this just as a local class in my application, I'll just paste it here verbatim for now:
```
using System;
using System.Collections.Generic;
using System.IO.Ports;
using System.IO;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using ESCPOS_NET;
using System.Runtime.InteropServices;
using System.Timers;

namespace kasse2 {
    public class WindowsPrinter : BasePrinter {
        private string _printerName;
        private MemoryStream _stream;

        public WindowsPrinter(string printerName) : base() {
            _stream = new MemoryStream();
            Writer = new BinaryWriter(_stream);
            _printerName = printerName;
        }

        public override void Flush(object sender, ElapsedEventArgs e) {
            if (BytesWrittenSinceLastFlush > 0) {
                var bytes = _stream.ToArray();
                _stream = new MemoryStream();
                Writer = new BinaryWriter(_stream);
                SendBytesToPrinter(_printerName, bytes);

            }
            BytesWrittenSinceLastFlush = 0;
        }

        [StructLayout(LayoutKind.Sequential, CharSet = CharSet.Ansi)]
        public class DOCINFOA {
            [MarshalAs(UnmanagedType.LPStr)] public string pDocName;
            [MarshalAs(UnmanagedType.LPStr)] public string pOutputFile;
            [MarshalAs(UnmanagedType.LPStr)] public string pDataType;
        }

        #region Declaration Dll

        [DllImport("winspool.Drv", EntryPoint = "OpenPrinterA", SetLastError = true, CharSet = CharSet.Ansi,
            ExactSpelling = true, CallingConvention = CallingConvention.StdCall)]
        public static extern bool OpenPrinter([MarshalAs(UnmanagedType.LPStr)] string szPrinter, out IntPtr hPrinter,
            IntPtr pd);

        [DllImport("winspool.Drv", EntryPoint = "ClosePrinter", SetLastError = true, ExactSpelling = true,
            CallingConvention = CallingConvention.StdCall)]
        public static extern bool ClosePrinter(IntPtr hPrinter);

        [DllImport("winspool.Drv", EntryPoint = "StartDocPrinterA", SetLastError = true, CharSet = CharSet.Ansi,
            ExactSpelling = true, CallingConvention = CallingConvention.StdCall)]
        public static extern bool StartDocPrinter(IntPtr hPrinter, int level,
            [In][MarshalAs(UnmanagedType.LPStruct)] DOCINFOA di);

        [DllImport("winspool.Drv", EntryPoint = "EndDocPrinter", SetLastError = true, ExactSpelling = true,
            CallingConvention = CallingConvention.StdCall)]
        public static extern bool EndDocPrinter(IntPtr hPrinter);

        [DllImport("winspool.Drv", EntryPoint = "StartPagePrinter", SetLastError = true, ExactSpelling = true,
            CallingConvention = CallingConvention.StdCall)]
        public static extern bool StartPagePrinter(IntPtr hPrinter);

        [DllImport("winspool.Drv", EntryPoint = "EndPagePrinter", SetLastError = true, ExactSpelling = true,
            CallingConvention = CallingConvention.StdCall)]
[... 90 lines cut ...]
            // Send the converted ANSI string to the printer.
            var result = SendBytesToPrinter(szPrinterName, pBytes, dwCount);
            Marshal.FreeCoTaskMem(pBytes);

            return result;
        }

        //if you want a wrapper function for you strings :
        public static bool SendAsciiToPrinter(string szPrinterName, string data) {
            var retval = false;

            //if  you are using UTF-8 and get wrong values in qrcode printing, you must use ASCII instead.
            //retval = SendBytesToPrinter(szPrinterName, Encoding.UTF8.GetBytes(data));
            retval = SendBytesToPrinter(szPrinterName, Encoding.ASCII.GetBytes(data));

            return retval;
        }

        #endregion
    }
}

```

This is basically mostly copied from that other project I mentioned, since it uses the same license we can at least talk with full license compatibility here.

reislanes, 2023-06-17:
I was researching the same thing. @sinni800 have you foked? Or made some personal implementation for your own?
As a fast solution i'm using both libraries, this one for Ethernet or Virtual COM printers, and the other one for USB printers.

lukevp, 2023-06-17:
I'm definitely interested in getting changes upstreamed in this repo to allow the windows print spooler, as long as it works well and it follows a similar API to the other printers, and properly indicates to the end user if their platform is not supported.   Honestly on MacOS and Linux, USB printers usually mount a file under `/dev/ttyUSBx` that you can print to without any drivers, so there's already a solution for USB printing from other platforms in most cases, so this could just be solved by documentation and the underlying implementation throwing an exception on Windows.

The problem I see in the code above is that it has implementation-specific methods for sending things to the printer. It would make more sense if the printer name / port / etc. were all set up in the constructor, and you just used it like any other printer implementation (construct an instance and then start writing bytes to it).

sinni800, 2023-06-19:
They are though, you only need to instantiate WindowsPrinter with a printer Name as the parameter. To print to it, you can use the BasePrinter methods. It could probably be cleaned up a bit, but it already works like that with the current code

reislanes, 2023-06-28:
Yeah, i first used it at my project as Elgin I9 printers use this implementation at the sample code guide to C# printing

The Class Helper i9 has the base code:

Imports System.Text
Imports System.Runtime.InteropServices
Imports System.IO
Namespace HelperI9
    Friend Class RawPrinterHelper
        ' Structure and API declarions:
        <StructLayout(LayoutKind.Sequential, CharSet:=CharSet.Ansi)>
        Public Class DOCINFOA
            <MarshalAs(UnmanagedType.LPStr)>
            Public pDocName As String
            <MarshalAs(UnmanagedType.LPStr)>
            Public pOutputFile As String
            <MarshalAs(UnmanagedType.LPStr)>
            Public pDataType As String
        End Class

        <DllImport("winspool.Drv", EntryPoint:="OpenPrinterA", SetLastError:=True, CharSet:=CharSet.Ansi, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function OpenPrinter(
        <MarshalAs(UnmanagedType.LPStr)> ByVal szPrinter As String, <Out> ByRef hPrinter As IntPtr, ByVal pd As IntPtr) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="ClosePrinter", SetLastError:=True, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function ClosePrinter(ByVal hPrinter As IntPtr) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="StartDocPrinterA", SetLastError:=True, CharSet:=CharSet.Ansi, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function StartDocPrinter(ByVal hPrinter As IntPtr, ByVal level As Integer,
        <[In], MarshalAs(UnmanagedType.LPStruct)> ByVal di As DOCINFOA) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="EndDocPrinter", SetLastError:=True, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function EndDocPrinter(ByVal hPrinter As IntPtr) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="StartPagePrinter", SetLastError:=True, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function StartPagePrinter(ByVal hPrinter As IntPtr) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="EndPagePrinter", SetLastError:=True, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function EndPagePrinter(ByVal hPrinter As IntPtr) As Boolean
        End Function

        <DllImport("winspool.Drv", EntryPoint:="WritePrinter", SetLastError:=True, ExactSpelling:=True, CallingConvention:=CallingConvention.StdCall)>
        Public Shared Function WritePrinter(ByVal hPrinter As IntPtr, ByVal pBytes As IntPtr, ByVal dwCount As Integer, <Out> ByRef dwWritten As Integer) As Boolean
        End Function

        ' SendBytesToPrinter()
        ' When the function is given a printer name and an unmanaged array
        ' of bytes, the function sends those bytes to the printer queue.
        ' Returns true on success, false on failure.
        Public Shared Function SendBytesToPrinter(ByVal szPrinterName As String, ByVal pBytes As IntPtr, ByVal dwCount As Integer) As Boolean
            Dim dwError = 0, dwWritten = 0
            Dim hPrinter As IntPtr = New IntPtr(0)
            Dim di As DOCINFOA = New DOCINFOA()
            Dim bSuccess = False ' Assume failure unless you specifically succeed.
            di.pDocName = "i9 RAW Document"
            di.pDataType = "RAW"

            ' Open the printer.
            If OpenPrinter(szPrinterName.Normalize(), hPrinter, IntPtr.Zero) Then
                ' Start a document.
                If StartDocPrinter(hPrinter, 1, di) Then
                    ' Start a page.
                    If StartPagePrinter(hPrinter) Then
                        ' Write your bytes.
                        bSuccess = WritePrinter(hPrinter, pBytes, dwCount, dwWritten)
[... 52 lines cut ...]
            Dim pBytes As IntPtr
            Dim dwCount As Integer
            ' How many characters are in the string?
            dwCount = szString.Length
            ' Assume that the printer is expecting ANSI text, and then convert
            ' the string to ANSI text.
            pBytes = Marshal.StringToCoTaskMemAnsi(szString)
            ' Send the converted ANSI string to the printer.
            SendBytesToPrinter(szPrinterName, pBytes, dwCount)
            Marshal.FreeCoTaskMem(pBytes)
            Return True
        End Function

        'if you want a wrapper function for you strings :
        Public Shared Function SendASCiiToPrinter(ByVal szPrinterName As String, ByVal data As String) As Boolean
            Dim retval = False

            'if  you are using UTF-8 and get wrong values in qrcode printing, you must use ASCII instead.
            'retval = SendBytesToPrinter(szPrinterName, Encoding.UTF8.GetBytes(data));

            retval = SendBytesToPrinter(szPrinterName, Encoding.ASCII.GetBytes(data))
            Return retval
        End Function
    End Class
End Namespace

matthiasvd94, 2023-11-08:
I have searched a lot to find a compatible COM port emulation and for me the Epson TM Virtual Port Driver is the only one that works properly for every chinese POS printer.
The standard driver for the device is a POS-58 driver which will not create a virtual COM but only a USB001 for example which cannot be used.
So in fact if you use the Epson TM you can connect every printer.