lukevp/ESC-POS-.NET #230: No existing code for BasePrinter.Startmonitoring()

Jury-jpg, 2023-05-21:


lukevp, 2023-06-17:
Hey @Jury-jpg  could you provide some more details on what your question / concern is regarding this function? Are you not sure how to use it? There is an example in the README:
```
// In your program, register event handler to call the method when printer status changes:
printer.StatusChanged += StatusChanged;

// and start monitoring for changes.
printer.StartMonitoring();
```

HGCollier, 2023-06-22:
`NetworkPrinter` does not contain a definition for `StartMonitoring`

wzuqui, 2024-03-06:
Looking at the tests implemented in this  [git](https://github.com/lukevp/ESC-POS-.NET/blob/035beec81a64442ed60c1769bd6155f1fed4b6dc/ESCPOS_NET.ConsoleTest/Program.cs#L248), I found how to get the data, maybe it will help you

```csharp
using(var xPrinter = new SerialPrinter(portName: "COM2", baudRate: 115200))
{
        xPrinter.StatusChanged += StatusChanged;
        xPrinter.Connected += ConnectedChanged;
        var xEpson = new EPSON();
        
        xPrinter.Write(xEpson.Initialize());
        xPrinter.Write(xEpson.Enable());

        xPrinter.Write(xEpson.EnableAutomaticStatusBack());
       Thread.Sleep(5000);
}
void StatusChanged(object sender, EventArgs e)
{
	var status = (PrinterStatusEventArgs)e;
	status.Dump();
}
```