lukevp/ESC-POS-.NET #233: EPSON TM-U220 Issue

jupagose, 2023-06-27:
I use  [ESC-POS-.NET) primary with thermal printers but some users still uses imact printers such  TM-U220 It prints but images does not print properly and each printing lines are split into two lines.

Definitelly must I don't use [ESC-POS-.NET] with this printers?

lukevp, 2023-06-27:
I'm familiar with impact printers but haven't tried the library with them. I believe it should work though. Try legacy image printing mode. Try the console.test app included with the source and see what does and doesn't work and report back!

jupagose, 2023-06-27:
Hi, I tryed adjusting width from 48 t0 40 chars and now it prints properly the text. But still having problems with logo.

Yes I tryed legacy mode, and when I change isHiDPI to false logo does not print!!!

jupagose, 2023-06-28:
Can I print flashed logo with ESC-POS.NET?

mtch99, 2023-07-10:
What do you mean by flash logo?  Provide more information and I will try best to help you

jupagose, 2023-07-10:
Epson printers has a feature that allow me to store the logo on it's flash memory 

https://www.epson-biz.com/modules/pos/index.php?page=single_soft&cid=2346

This allow me whe use the driver to config if it prints before / after any print job. but with oPos does not work .

My question is about if ican send an oPos command to force the printer to print this logo, because with the TM-U220 model logo does not print properly.

mtch99, 2023-07-13:
I will check it out and come back to you by next Thursday