lukevp/ESC-POS-.NET #236: Label Printing Support?

monsterkane05, 2023-07-14:
Hi,

I was wondering if there is any planned support for label printers which use ESC/POS such as the Epson TM-L90. mainly the feed to label mark, and feed to cutter commands.  this library can be used to print to them but labels end up being out of line due to them not being able to start at the top of the label.

[Epson Documentation](https://files.support.epson.com/pdf/pos/bulk/esclabel_crg_en_07.pdf )

lukevp, 2023-07-18:
Hey @monsterkane05 !

This repo is open to PRs, and it's super easy to add new command sets.  You should be able to follow the same conventions as the other commands to add these in.  Are you interested in contributing that? Would be happy to merge and re-release the nuget package with this, as it seems like a straightforward change.

I've been focusing on supporting the most commonly-used commands in favor of supporting every command under the sun, so if there are only a couple really useful ones in this command set then I'm good with just doing those.

monsterkane05, 2023-07-18:
Yeah im happy to give it a try, reas

closed 2023-07-18 by monsterkane05