lukevp/ESC-POS-.NET #241: Not getting status from SerialPrinter

NathanTe, 2023-08-18:
I'm not receiving any status updates from my printer connected through serial. 

I have tried : 
- using the `Status` property => doesn't update
- using the `StatusChanged` event => doesn't fire

I have found some answers in issues where you say to use `StartMonitoring()` but this method doesn't exist on `SerialPrinter` class. 
Is this method still required? if so where can I find it?

lukevp, 2024-01-14:
Yes, you need to enable monitoring or the printer will not send status updates back down the wire to your .NET app.  It's a specific setting within the printer that has to be enabled.

It looks like the 3.0 release does not have support for status monitoring (some of it is commented out).  You could try the previous releases and see if those work for you, and carry forward the status code into the latest version. it should be fairly easy to add.  I think the reason that it was removed from 3.0 is because not all of the printer types support it, and we were trying to get the immediate network printer support stabilized.  The status monitoring worked great in past versions so it should be able to be re-added.  Let me know if you're interested in working on this, would love to have it working again.  My guess is that we'll need to shift some of the code into the SerialPrinter class and have the BasePrinter provide overrides for the implementations in the subclasses.

rokka, 2026-01-09:
I am using 3.0.0 and what helps in my case is the following:

```
serialPrinter = new SerialPrinter(portName: comPort, baudRate: baudRate);

ICommandEmitter e = new EPSON();
serialPrinter.Write(e.Initialize());
serialPrinter.Write(e.Enable());
serialPrinter.Write(e.EnableAutomaticStatusBack());

var status = serialPrinter.Status;
...
```