lukevp/ESC-POS-.NET #242: Can't Print Greek characters

ChrisONZ, 2023-08-26:
Hi,

I'm trying to create a receipt for a cafe, and will need to print Greek characters. Using your testing code, i have "cribbed" the following

       public static byte[][] Receipt(ICommandEmitter e, string ItemDetails) => new byte[][] {
           e.CodePage(CodePage.PC737_GREEK),
           e.CenterAlign(),
           e.PrintLine(),
           e.PrintImage(File.ReadAllBytes("C:/Users/cwowe/OneDrive/Documents/CashRegisterApplication/application/receipt-logo.png"), true),
           //e.PrintLine(),
           //e.SetBarcodeHeightInDots(360),
           //e.SetBarWidth(BarWidth.Default),
           //e.SetBarLabelPosition(BarLabelPrintPosition.None),
           //e.PrintBarcode(BarcodeType.ITF, "0123456789"),
           e.PrintLine(),
           e.PrintLine("MIGNIARDISE"),
           e.PrintLine("Veikou 30, Koukaki"),
           e.PrintLine("Athens 117 42"),
           e.PrintLine("21 1750 2334 / 69 7484 0903"),
           e.SetStyles(PrintStyle.Underline),
           e.PrintLine("www.migniardise.gr"),
           e.SetStyles(PrintStyle.None),
           e.PrintLine(),
           e.LeftAlign(),
           e.PrintLine("Receipt: a5b8b76c-ee2e-4cfe-85e1-3942d24da042"),
           e.PrintLine("Date: 25/08/2023      Time 11:54"),
           e.PrintLine(),
           e.PrintLine(),
           //e.SetStyles(PrintStyle.FontB),
           e.PrintLine("Qty   Item                       price       tax"),
           e.PrintLine("------------------------------------------------"),
           e.PrintLine(ItemDetails),
           e.PrintLine(" 1   Croissant (βούτυρο)          1.60       13%"),
           e.PrintLine(" 2   Kιουνεφέ ζεστό              10.00       13%"),
           e.PrintLine(" 2   Μπύρα                       10.00       24%"),
           e.PrintLine("------------------------------------------------"),
           e.RightAlign(),
           //e.PrintLine("SUBTOTAL         21.60"),
           e.SetStyles(PrintStyle.Bold),
           e.PrintLine("      Total:         21.60"),
           e.PrintLine(),
           e.CenterAlign(),
           e.PrintLine("***  THANK YOU FOR YOUR CUSTOM  ***"),
           e.SetStyles(PrintStyle.None),
           e.PrintLine(),
           e.PrintLine(),
           e.Print2DCode(TwoDimensionCodeType.QRCODE_MODEL1, "www.mfa.gr/en/index.html", Size2DCode.LARGE),
           e.PrintLine(),
           e.PrintLine()
       };
   }

using 737Greek from the vendors testing app (Its a Munbyn ITPP047P thermal) it all prints fine and i can see the character set is Greek. 
[image]

but when i print using the above code, i get the following output - in the receipt lines, any Greek characters are replaced with odd symbols...
[image]

am I just using it incorrectly or could there be a problem, if the former, does anyone have an example of what i'm trying to do? I also note that just printing the greek characters from Notepad++ directly works

any help would be very much appreciated - many thanks in advance!

igorocampos, 2023-09-01:
Hey, I don't have a printer with me, so I can't reproduce your issue, but in case you haven't read these already, they might give you some insight on the issue:

https://github.com/lukevp/ESC-POS-.NET/issues/67 https://github.com/lukevp/ESC-POS-.NET/issues/77 https://github.com/lukevp/ESC-POS-.NET/issues/86 https://github.com/lukevp/ESC-POS-.NET/issues/88 https://github.com/lukevp/ESC-POS-.NET/issues/103

ChrisONZ, 2023-09-02:
Hey, thanks for the direction, one of the examples helped.

the key was to use "ibm737" to represent the same as the Greek737 on the printer.... i had tried the "ISO-8859-7" page name previously

was then able to just pass the byte array(s) to my main printer routines...

```
            foreach (RegisterSaleItem item in lstReceiptLines)
            {
                ItemDetails = string.Format("{0,2:D}  {1,-25}{2,8}{3,9}%", item.SaleQuantity, item.ItemDesc, item.SaleTotalValue.ToString("##0.00"), item.VatPercent.ToString("#0.00"));
                byte[] GreekBytes = Encoding.GetEncoding("ibm737").GetBytes(ItemDetails);
                _printer.Write(ReceiptComponents.ReceiptLine(_ea, GreekBytes));
            }
```

and use the code example

```
        public static byte[][] ReceiptLine(ICommandEmitter e, byte[] LineDetails) => new byte[][] {
            e.CodePage(CodePage.PC737_GREEK),
            LineDetails,
            e.PrintLine()

        };
```

still cant print a Euro symbol, but that's not really needed for this project ( i did check your page about printing euro but it seems to leave my printer in the codpage for Euro and doesnt return back to the greek codepage....), im sure I'll figure it out if needed.

Thanks a lot for your pointers!

closed 2023-09-02 by ChrisONZ