lukevp/ESC-POS-.NET #247: ESC-POS-.NET want work on MAUI Android - issue libSystem.IO.Ports.Native not loaded

bokidotnet, 2023-11-03:
I'm getting exception like bellow when I want to use SerialPrinter class object.

Issue is present on both .net 6 and .net 7 based MAUI Android projects.

runtime.native.System.IO.Ports package is present in dependencies (give a shot and update to version 7.0.0 but same issue).

Any way is it possible to use this lib on MAUI Android based devices and how? e.g. for printing on BT POS printer device?

`PrinterConnect(printerName, printerBTAddress);`

`    private bool PrinterConnect(string printerName, string printerBTAddress)
    {

        try
        {

            var printer = new SerialPrinter(portName: printerName, baudRate: 115200);

            return true;

        }
        catch (Exception ex)
        {

            Debug.WriteLine($"---> PrinterConnect() exception: {ex}");

        }

        return false;

    }`

Exception
[0:] ---> PrinterConnect() exception: System.DllNotFoundException: libSystem.IO.Ports.Native
   at System.IO.Ports.SafeSerialDeviceHandle.Open(String portName)
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
   at ESCPOS_NET.SerialPrinter..ctor(String portName, Int32 baudRate)
   at ESCPOSMauiTestApp.MainPage.PrinterConnect(String printerName, String printerBTAddress) in E:\projects\testing\dotnet\MAUI\ESCPOSMauiTestApp\ESCPOSMauiTestApp\MainPage.xaml.cs:line 29

bokidotnet, 2023-11-07:
I solved problem by using native android BT lib and his socket to write bytes directly to socket and used Epson class as helper to create commands and payload data to write on opened socket.

closed 2023-12-05 by igorocampos

igorocampos, 2023-12-05:
described solution is related to #237